fl4p/batmon-ha #217: Password for JBD BMS

AstroR44, 2024-04-02:
The newer JBD-BMS require a PIN or password. Even if this is specified in the configuration in the same way as for victron, access to the BMS is not possible. 

Is there an option or will there be an update?

JBD BMS 4S 150A +Bluetooth

LifeofDan-EL, 2024-08-30:
Hello, have you been able to get it to connect?

AstroR44, 2024-09-02:
No, I did not succeed.

In the meantime, I have tried another independent HA instance and have had no success with that either.
Without a PIN, the attempt ends with a timeout. With PIN I get the error message:

11:55:39 ERROR [sampling]  XXXXXXX error (#3): [org.bluez.Error.AuthenticationFailed] Authentication Failed

LifeofDan-EL, 2024-09-03:
I see, I thought this was for the JK BMS. I found out that this addon was able to connect without adding the pin. However, the pin is the default "1234" paring pin.

fl4p, 2026-09-03:
Newer JBD firmware does not use the app password frame at all, it requires a proper passkey BLE pairing and then answers the normal protocol (traced in syssi/esphome-jbd-bms#245). batmon does that with `pin: "123456"` next to `type: jbd`; in 2024 it failed because the bleak of that time had no pairing agent, that has been replaced since. Docs updated in 7631756. Not tested on such a pack myself, so if it still says AuthenticationFailed on a current version, open a new issue with a verbose_log.

closed 2026-09-03 by fl4p