fl4p/batmon-ha #219: Add-on breaks quite often.

LifeofDan-EL, 2024-04-10:
Hello, I find this add-on very useful and I appreciate the work you are doing. I have this issue that seems to pop up quite often, where the Add-on timeouts during sampling and even stop working after it has been working for a while. I use the JBD-SP16S020-L8S 

```
Paste log output between BEGIN and END:
BEGIN 
21:06:12 DEBUG [manager] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_DD_6D_73_D1_05_01): ['org.bluez.Device1', {'RSSI': <dbus_fast.signature.Variant ('n', -40)>}, []]
21:06:12 DEBUG [manager] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_6C_EE_61_81_05_8F): ['org.bluez.Device1', {'RSSI': <dbus_fast.signature.Variant ('n', -52)>}, []]
21:06:13 DEBUG [manager] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_00_DE_B5_40_E1_29): ['org.bluez.Device1', {'RSSI': <dbus_fast.signature.Variant ('n', -57)>}, []]
21:06:13 DEBUG [manager] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_6C_EE_61_81_05_8F): ['org.bluez.Device1', {'RSSI': <dbus_fast.signature.Variant ('n', -52)>}, []]
21:06:13 DEBUG [manager] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_00_DE_B5_40_E1_29): ['org.bluez.Device1', {'RSSI': <dbus_fast.signature.Variant ('n', -64)>}, []]
21:06:13 DEBUG [manager] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_6C_EE_61_81_05_8F): ['org.bluez.Device1', {'RSSI': <dbus_fast.signature.Variant ('n', -47)>}, []]
21:06:13 DEBUG [manager] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_00_DE_B5_40_E1_29): ['org.bluez.Device1', {'RSSI': <dbus_fast.signature.Variant ('n', -58)>}, []]
21:06:13 DEBUG [manager] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_6C_EE_61_81_05_8F): ['org.bluez.Device1', {'RSSI': <dbus_fast.signature.Variant ('n', -46)>}, []]
21:06:14 DEBUG [manager] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_6C_EE_61_81_05_8F): ['org.bluez.Device1', {'RSSI': <dbus_fast.signature.Variant ('n', -40)>}, []]
21:06:14 DEBUG [manager] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_C4_82_E1_B4_41_AF): ['org.bluez.Device1', {}, ['RSSI']]
21:06:14 DEBUG [manager] received D-Bus signal: org.freedesktop.DBus.ObjectManager.InterfacesRemoved (/): ['/org/bluez/hci0/dev_C4_82_E1_B4_41_AF', ['org.freedesktop.DBus.Properties', 'org.freedesktop.DBus.Introspectable', 'org.bluez.Device1']]
21:06:14 DEBUG [manager] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_00_DE_B5_40_E1_29): ['org.bluez.Device1', {}, ['RSSI']]
21:06:14 DEBUG [manager] received D-Bus signal: org.freedesktop.DBus.ObjectManager.InterfacesRemoved (/): ['/org/bluez/hci0/dev_00_DE_B5_40_E1_29', ['org.freedesktop.DBus.Properties', 'org.freedesktop.DBus.Introspectable', 'org.bluez.Device1']]
21:06:14 DEBUG [manager] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_DD_6D_73_D1_05_01): ['org.bluez.Device1', {}, ['RSSI']]
21:06:14 DEBUG [manager] received D-Bus signal: org.freedesktop.DBus.ObjectManager.InterfacesRemoved (/): ['/org/bluez/hci0/dev_DD_6D_73_D1_05_01', ['org.freedesktop.DBus.Properties', 'org.freedesktop.DBus.Introspectable', 'org.bluez.Device1']]
21:06:14 DEBUG [manager] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_CB_4E_FC_7C_48_83): ['org.bluez.Device1', {}, ['RSSI']]
21:06:14 DEBUG [manager] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_6C_EE_61_81_05_8F): ['org.bluez.Device1', {}, ['TxPower', 'RSSI']]
21:06:14 DEBUG [manager] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0): ['org.bluez.Adapter1', {'Discovering': <dbus_fast.signature.Variant ('b', False)>}, []]
21:06:14 INFO [bt] BT 6C:EE:61:81:05:8F          6C-EE-61-81-05-8F
21:06:14 INFO [bt] BT C4:82:E1:B4:41:AF                         TY
21:06:14 INFO [bt] BT 00:DE:B5:40:E1:29          00-DE-B5-40-E1-29
21:06:14 INFO [bt] BT CB:4E:FC:7C:48:83          CB-4E-FC-7C-48-83
21:06:14 INFO [bt] BT DD:6D:73:D1:05:01          DD-6D-73-D1-05-01
21:06:14 ERROR [sampling] JbdBt(70:3E:97:EB:2E:B7,battery1) device not found, retry in 7 seconds
21:06:15 DEBUG [sampling] retry in 7 sec
21:06:20 DEBUG [sampling] retry in 2 sec
21:06:25 INFO [sampling] connecting bms JbdBt(70:3E:97:EB:2E:B7,battery1)
21:06:25 INFO [bt] connecting battery1 (70:3E:97:EB:2E:B7) adapter=default timeout=20
21:06:25 DEBUG [client] Connecting to device @ 70:3E:97:EB:2E:B7
21:06:25 DEBUG [manager] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0): ['org.bluez.Adapter1', {'Discovering': <dbus_fast.signature.Variant ('b', True)>}, []]
21:06:25 DEBUG [manager] received D-Bus signal: org.freedesktop.DBus.ObjectManager.InterfacesAdded (/): ['/org/bluez/hci0/dev_00_DE_B5_40_E1_29', {'org.freedesktop.DBus.Introspectable': {}, 'org.bluez.Device1': {'Address': <dbus_fast.signature.Variant ('s', 00:DE:B5:40:E1:29)>, 'AddressType': <dbus_fast.signature.Variant ('s', random)>, 'Alias': <dbus_fast.signature.Variant ('s', 00-DE-B5-40-E1-29)>, 'Paired': <dbus_fast.signature.Variant ('b', False)>, 'Bonded': <dbus_fast.signature.Variant ('b', False)>, 'Trusted': <dbus_fast.signature.Variant ('b', False)>, 'Blocked': <dbus_fast.signature.Variant ('b', False)>, 'LegacyPairing': <dbus_fast.signature.Variant ('b', False)>, 'RSSI': <dbus_fast.signature.Variant ('n', -68)>, 'Connected': <dbus_fast.signature.Variant ('b', False)>, 'UUIDs': <dbus_fast.signature.Variant ('as', [])>, 'Adapter': <dbus_fast.signature.Variant ('o', /org/bluez/hci0)>, 'ManufacturerData': <dbus_fast.signature.Variant ('a{qv}', {6: <dbus_fast.signature.Variant ('ay', bytearray(b'\x01\t "\x81\xcf\xeb\xfe\xbc\xd3k\x82\x07<\xd9!Y\xa0\x91m\xb0\xb2\xb6\x8e$\xdd\x89'))>})>, 'ServicesResolved': <dbus_fast.signature.Variant ('b', False)>, 'AdvertisingFlags': <dbus_fast.signature.Variant ('ay', bytearray(b'\x00'))>}, 'org.freedesktop.DBus.Properties': {}}]
21:06:25 DEBUG [manager] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_00_DE_B5_40_E1_29): ['org.bluez.Device1', {'RSSI': <dbus_fast.signature.Variant ('n', -67)>}, []]
21:06:25 DEBUG [manager] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_6C_EE_61_81_05_8F): ['org.bluez.Device1', {'RSSI': <dbus_fast.signature.Variant ('n', -55)>, 'TxPower': <dbus_fast.signature.Variant ('n', 7)>}, []]
21:06:25 DEBUG [manager] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_00_DE_B5_40_E1_29): ['org.bluez.Device1', {'RSSI': <dbus_fast.signature.Variant ('n', -67)>}, []]
21:06:25 DEBUG [manager] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_6C_EE_61_81_05_8F): ['org.bluez.Device1', {'RSSI': <dbus_fast.signature.Variant ('n', -40)>}, []]
21:06:25 DEBUG [manager] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_00_DE_B5_40_E1_29): ['org.bluez.Device1', {'RSSI': <dbus_fast.signature.Variant ('n', -54)>}, []]
21:06:25 DEBUG [manager] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_00_DE_B5_40_E1_29): ['org.bluez.Device1', {'RSSI': <dbus_fast.signature.Variant ('n', -59)>}, []]
21:06:26 DEBUG [manager] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_6C_EE_61_81_05_8F): ['org.bluez.Device1', {'RSSI': <dbus_fast.signature.Variant ('n', -46)>}, []]
21:06:26 DEBUG [manager] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_00_DE_B5_40_E1_29): ['org.bluez.Device1', {'RSSI': <dbus_fast.signature.Variant ('n', -53)>}, []]
21:06:26 DEBUG [manager] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_00_DE_B5_40_E1_29): ['org.bluez.Device1', {'RSSI': <dbus_fast.signature.Variant ('n', -54)>}, []]
[... cut ...]
21:06:35 DEBUG [manager] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_CB_4E_FC_7C_48_83): ['org.bluez.Device1', {'RSSI': <dbus_fast.signature.Variant ('n', -60)>}, []]
21:06:35 DEBUG [manager] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_00_DE_B5_40_E1_29): ['org.bluez.Device1', {'RSSI': <dbus_fast.signature.Variant ('n', -57)>}, []]
21:06:35 DEBUG [manager] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_6C_EE_61_81_05_8F): ['org.bluez.Device1', {'RSSI': <dbus_fast.signature.Variant ('n', -52)>}, []]
END
```

LifeofDan-EL, 2024-04-10:
Without Verbose
```
File "/app/bmslib/bt.py", line 237, in connect
    await self._connect_client(timeout=timeout)
  File "/app/bmslib/bt.py", line 186, in _connect_client
    await asyncio.wait_for(self.client.connect(timeout=timeout), timeout=timeout + 1)
  File "/usr/lib/python3.11/asyncio/tasks.py", line 489, in wait_for
    return fut.result()
           ^^^^^^^^^^^^
  File "/app/venv/lib/python3.11/site-packages/bleak/__init__.py", line 531, in connect
    return await self._backend.connect(**kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/venv/lib/python3.11/site-packages/bleak/backends/bluezdbus/client.py", line 141, in connect
    async with async_timeout(timeout):
  File "/usr/lib/python3.11/asyncio/timeouts.py", line 115, in __aexit__
    raise TimeoutError from exc_val
TimeoutError
21:40:49 INFO [sampling] Bleak version 0.20.2
21:40:49 INFO [sampling] connecting bms JbdBt(70:3E:97:EB:2E:B7,battery1)
21:41:09 ERROR [sampling] battery1 error (#2): <class 'TimeoutError'>
Traceback (most recent call last):
  File "/app/venv/lib/python3.11/site-packages/bleak/backends/bluezdbus/client.py", line 204, in connect
    reply = await self._bus.call(
            ^^^^^^^^^^^^^^^^^^^^^
  File "/app/venv/lib/python3.11/site-packages/dbus_fast/aio/message_bus.py", line 390, in call
    await future
asyncio.exceptions.CancelledError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/app/bmslib/sampling.py", line 155, in __call__
    s = await self._sample_inner()
        ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/bmslib/sampling.py", line 227, in _sample_inner
    async with bms:
  File "/app/bmslib/bt.py", line 345, in __aenter__
    await self.connect()
  File "/app/bmslib/models/jbd.py", line 53, in connect
    await super().connect(**kwargs)
  File "/app/bmslib/bt.py", line 237, in connect
    await self._connect_client(timeout=timeout)
  File "/app/bmslib/bt.py", line 186, in _connect_client
    await asyncio.wait_for(self.client.connect(timeout=timeout), timeout=timeout + 1)
  File "/usr/lib/python3.11/asyncio/tasks.py", line 489, in wait_for
    return fut.result()
           ^^^^^^^^^^^^
  File "/app/venv/lib/python3.11/site-packages/bleak/__init__.py", line 531, in connect
    return await self._backend.connect(**kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/venv/lib/python3.11/site-packages/bleak/backends/bluezdbus/client.py", line 141, in connect
    async with async_timeout(timeout):
  File "/usr/lib/python3.11/asyncio/timeouts.py", line 115, in __aexit__
    raise TimeoutError from exc_val
TimeoutError
21:41:09 INFO [sampling] Bleak version 0.20.2
21:41:09 ERROR [main] 1 exceptions occurred fetching BMSs
21:41:09 ERROR [main] Error (num 1, max 200) reading BMS: 
21:41:09 ERROR [main] Stack: Traceback (most recent call last):
  File "/app/venv/lib/python3.11/site-packages/bleak/backends/bluezdbus/client.py", line 204, in connect
    reply = await self._bus.call(
            ^^^^^^^^^^^^^^^^^^^^^
  File "/app/venv/lib/python3.11/site-packages/dbus_fast/aio/message_bus.py", line 390, in call
    await future
asyncio.exceptions.CancelledError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/app/main.py", line 36, in fetch_loop
    if await fn():
       ^^^^^^^^^^
  File "/app/main.py", line 319, in fn
    raise exceptions[0]
  File "/app/main.py", line 314, in fn
    await t()
  File "/app/bmslib/sampling.py", line 155, in __call__
    s = await self._sample_inner()
        ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/bmslib/sampling.py", line 227, in _sample_inner
    async with bms:
  File "/app/bmslib/bt.py", line 345, in __aenter__
    await self.connect()
  File "/app/bmslib/models/jbd.py", line 53, in connect
    await super().connect(**kwargs)
  File "/app/bmslib/bt.py", line 237, in connect
    await self._connect_client(timeout=timeout)
  File "/app/bmslib/bt.py", line 186, in _connect_client
    await asyncio.wait_for(self.client.connect(timeout=timeout), timeout=timeout + 1)
  File "/usr/lib/python3.11/asyncio/tasks.py", line 489, in wait_for
    return fut.result()
           ^^^^^^^^^^^^
  File "/app/venv/lib/python3.11/site-packages/bleak/__init__.py", line 531, in connect
    return await self._backend.connect(**kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/venv/lib/python3.11/site-packages/bleak/backends/bluezdbus/client.py", line 141, in connect
    async with async_timeout(timeout):
  File "/usr/lib/python3.11/asyncio/timeouts.py", line 115, in __aexit__
    raise TimeoutError from exc_val
TimeoutError

21:41:10 INFO [sampling] connecting bms JbdBt(70:3E:97:EB:2E:B7,battery1)
```

fl4p, 2026-09-03:
The scan sees five neighbours but never the JBD module, so it was not advertising at that moment, and the connect timeout that follows is the same thing. The JBD module serves one connection and goes silent while the xiaoxiang app on a phone is connected or auto-reconnecting; close the app on every phone. Since this log (2024) the retry/keep-alive path has been rewritten several times and the `bluek` stack replaced BlueZ over D-Bus, so if it still drops on a current version open a new issue with a verbose_log around the dropout.

closed 2026-09-03 by fl4p