fl4p/batmon-ha #220: Trouble finding any BMS through ESPHome Bluetooth Proxy (SMLIGHT SLZB-06M) for remote location

Off-Grid-Garage, 2024-04-11:
Hello and thank you for your great integration and solution by connecting to many supported BMS via BT.

Background (and problem):
My HA installation and hardware is installed inside my house while my batteries and BMS are setup in my garage, a separate building ~50m away from the house. I have a network connection between the buildings and have several services running through that connection successfully. This is too far away to connect any BT devices though.

The Solution:
I recently bought and installed a [SMLIGHT SLZB-06M](https://smlight.tech/manual/slzb-06/guide/getting-started/) Zigbee, Wifi, Bluetooth Ethernet extender in my garage. The Zigbee over Ethernet Mode works a treat (adds a second coordinator to one HA installation).
[image]
The SLZB has now been flashed with [ESPHome Bluetooth Proxy](https://smlight.tech/manual/slzb-06/guide/bt-proxy/) firmware provided by SMLIGHT and it works fine with all my Bluetooth devices inside my garage (BT LED lights, Ruuvi Tags, ...) and I can control these BT devices now from inside my house.

The only think which does not work is batmon-HA and I'm struggling to find the right adapter name for this SLZB device. Batmon-HA uses hci0, hci1,... to specify the BT adapter. This obviously does not work as the SLZB is not a generic BT adapter attached to the HA hardware. BT discovery and scan only find devices in my house as it falls back to the default BT devises it knows about.

HA obviously has no trouble to connect to the remote BT devices through the ESPHome Proxy, so the question arises if batmon-HA is able to use this same BT proxy connection.

A lot of people would be interested in such a solution when their HA installation is not in BT range of their BMS. Any help and support to resolve this is much appreciated. Please let me know how I can assist with further information and testing.

Cheers and thank you again from sunny hot Australia
Andy

[image]
[image]
[image]

Off-Grid-Garage, 2024-04-26:
Hello @7wells !

No, I haven't unfortunately. I'm still unable to find the device name to add to the adapter in Batmon.

Would really like to get this resolved...

At the end, I can swap the HA server and the SMLIGHT BT Proxy (HA goes in the garage, SMLIGHT goes in my house). Having the server (with its 'genuine' BT hardware) near the BMSes will definitely work with the Batmon integration, but I would prefer not to swap the hardware just for this reason.

There must be a way to find the device name and add this to the Batmon integration. That is all we need...

patman15, 2024-05-28:
@7wells, @Off-Grid-Garage this addon does not support ESPHome bluetooth proxy. See https://github.com/fl4p/batmon-ha/discussions/213#discussion-6419834 for a possible solution, but jikong is not (yet) supported.

patman15, 2024-07-10:
Support added.

huskyte, 2024-08-26:
hmmm, and how does the solution look like then?

adapter: "Slzb..." or is it automatically choosen?

huskyte, 2024-08-26:
[image]

This is how my SLZB shows up. Do I use the name or the address?

patman15, 2024-08-31:
> hmmm, and how does the solution look like then?
> 
see https://github.com/fl4p/batmon-ha/discussions/213#discussion-6419834