fl4p/batmon-ha #226: Supervolt BMS | Error | Characteristic 6e400003-b5a3-f393-e0a9-e50e24dcca9e not found! | unsupported operand type(s) for &: 'NoneType' and 'int'

nightcat91, 2024-04-26:
2x Supervolt SX150P 

- address: C4:08:7D:7F:B8:1C
  type: supervolt
  alias: Vorne1
- address: E6:F9:20:27:3C:31
  type: supervolt
  alias: Hinten2

Error:
ERROR [supervolt] (<class 'bleak.exc.BleakError'>, BleakError('Characteristic 6e400002-b5a3-f393-e0a9-e50e24dcca9e was not found!')
ERROR [main] Error reading BMS: unsupported operand type(s) for &: 'NoneType' and 'int'

I already tried "install_newer_bleak", "bt_power_cycle" and with ot without "keep_alive"
Not quite sure how to get a complete log, down below are pieces of logs which I got from the addon logs tab with careful refreshing and copy...

If any more info is needed please let me know.
 

```
Paste log output between BEGIN and END:
BEGIN 

s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
19:01:53 INFO [install_bleak] bleak 0.13.1a1 installed, want 0.20.2, running pip3 install bleak==0.20.2
19:01:53 INFO [install_bleak] influxdb installed=False need=False
Looking in indexes: https://pypi.org/simple, https://wheels.home-assistant.io/musllinux-index/
Collecting bleak==0.20.2
  Downloading https://wheels.home-assistant.io/musllinux-index/bleak-0.20.2-py3-none-any.whl (135 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 135.6/135.6 kB 598.0 kB/s eta 0:00:00
Collecting dbus-fast<2.0.0,>=1.83.0 (from bleak==0.20.2)
  Downloading https://wheels.home-assistant.io/musllinux-index/dbus_fast-1.95.2-cp311-cp311-musllinux_1_2_aarch64.whl (1.7 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 1.7/1.7 MB 5.0 MB/s eta 0:00:00
Installing collected packages: dbus-fast, bleak
  Attempting uninstall: bleak
    Found existing installation: bleak 0.13.1a1
    Uninstalling bleak-0.13.1a1:
      Successfully uninstalled bleak-0.13.1a1
Successfully installed bleak-0.20.2 dbus-fast-1.95.2
19:02:03 INFO [main] Power cycle bluetooth hardware
Unable to open mgmt_socket
Unable to open mgmt_socket
19:02:06 INFO [bt] BT Discovery:
19:02:12 INFO [bt] BT E9:66:91:7B:15:C8                  Victron 1
19:02:12 INFO [bt] BT CE:A0:09:E7:49:9D     SmartShunt HQ2220VVMXM
19:02:12 INFO [bt] BT E0:1E:AF:76:E6:EE SmartBatterySense HQ21237KXFV
19:02:12 INFO [bt] BT C7:19:54:F9:0A:3B                  Victron 2
19:02:12 INFO [bt] BT 42:8B:15:75:B7:A7         Truma iNetX-2F3DBD
19:02:12 INFO [bt] BT C4:08:7D:7F:B8:1C             SX150P-B230101
19:02:12 INFO [bt] BT E6:F9:20:27:3C:31             SX150P-B230149
19:02:12 INFO [bt] BT 7B:43:6E:AF:CB:18           Jabra Evolve2 85
19:02:12 INFO [bt] BT 10:D3:42:27:2F:B1          10-D3-42-27-2F-B1
19:02:12 INFO [bt] BT 4D:85:DD:0B:F3:31           Jabra Evolve 65e
19:02:12 INFO [bt] BT 55:5C:7D:C1:91:32               JBL Charge 5
19:02:12 INFO [bt] BT 69:73:8B:1B:DA:D5          69-73-8B-1B-DA-D5
19:02:12 INFO [bt] BT D9:11:A9:01:5D:85          D9-11-A9-01-5D-85
19:02:12 INFO [bt] BT 4E:AF:AF:EC:91:17          4E-AF-AF-EC-91-17
19:02:12 INFO [bt] BT 4C:18:E6:33:D3:0C          4C-18-E6-33-D3-0C
19:02:12 INFO [main] Verbose logging enabled
19:02:12 INFO [main] Bleak version 0.20.2, BtBackend version bluez-v5.70
19:02:12 DEBUG [bt] BMS SuperVoltBt(C4:08:7D:7F:B8:1C,Vorne1) keep alive enabled
19:02:12 DEBUG [bt] BMS SuperVoltBt(E6:F9:20:27:3C:31,Hinten2) keep alive enabled
19:02:12 INFO [main] connecting mqtt addons@core-mosquitto
[... 90 lines cut ...]
  File "/app/venv/lib/python3.11/site-packages/bleak/__init__.py", line 659, in write_gatt_char
    await self._backend.write_gatt_char(char_specifier, data, response)
  File "/app/venv/lib/python3.11/site-packages/bleak/backends/bluezdbus/client.py", line 848, in write_gatt_char
    raise BleakError("Characteristic {0} was not found!".format(char_specifier))
bleak.exc.BleakError: Characteristic 6e400002-b5a3-f393-e0a9-e50e24dcca9e was not found!
19:02:25 ERROR [sampling] Hinten2 error (#9): unsupported operand type(s) for &: 'NoneType' and 'int'
Traceback (most recent call last):
  File "/app/bmslib/sampling.py", line 155, in __call__
    s = await self._sample_inner()
        ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/bmslib/sampling.py", line 237, in _sample_inner
    sample = await bms.fetch()
             ^^^^^^^^^^^^^^^^^
  File "/app/bmslib/models/supervolt.py", line 397, in fetch
    status_discharging=(self.workingState & 0x0002 > 0),
                        ~~~~~~~~~~~~~~~~~~^~~~~~~~
TypeError: unsupported operand type(s) for &: 'NoneType' and 'int'
19:02:25 INFO [sampling] Bleak version 0.20.2

###########################
Additional Log Entry from home-assistant.log:
2024-04-26 19:03:56.869 ERROR (MainThread) [habluetooth.scanner] hci0 (00:1A:7D:DA:71:13): Error stopping scanner: [org.bluez.Error.Failed] No discovery started

END
```

tinkercasa, 2024-05-15:
I get similar errors.

ERROR:
```
bleak.exc.BleakError: Characteristic 6e400002-b5a3-f393-e0a9-e50e24dcca9e was not found!
[..]
status_discharging=(self.workingState & 0x0002 > 0),
                        ~~~~~~~~~~~~~~~~~~^~~~~~~~
TypeError: unsupported operand type(s) for &: 'NoneType' and 'int'
```

HA Core: 2024.5.3
HA Supervisor: 2024.05.1
HA OS: 12.3
Bluetooth: Home Assistant Yellow / Raspberry Pi Compute Module 4
Battery: Supervolt SX-100 (**2024 version**)
Batmon Version: 1.78
Batmon Configuration:
```
- address: XX:XX:XX:XX:XX:XX
  type: supervolt
  alias: battery_sx100
  verbose: true
```

Without knowing any better I suspect that Batmons supervolt type needs to be updated for the 2024 version of the batteries. The new batteries have a blue top and the old batteries have a black top. The new 2024 version might have changes in the BMS /  BLE communication.

Some more information just to show that the battery is visible via BLE and is generally accessible:

The BMS is accessible via the Supervolt companion app.
When searching for the battery with the ESPHome Bluetooth Low Energy Tracker the battery shows up:

address: XX:XX:XX:XX:XX:XX
name: SX100-B230019
advertised service UUID: 0xFF00
advertised service data: \<empty\>
advertised manufacturer data: 0x007B

According to [Bluetooth-Company-Identifiers.csv](https://gist.github.com/angorb/f92f76108b98bb0d81c74f60671e9c67#file-bluetooth-company-identifiers-csv) **0x007B** stands for **Hanlynn Technologies**. I could confirm this with the android app BLE Radar.

### How to get the complete batmon add-on log
  * install add-on **Terminal & SSH** (if you haven't already)
  * open **Settings -> Add-ons -> Terminal & SSH -> OPEN WEB UI**
  * execute the following command in the terminal and copy the identifier shown:<br/>`ha host logs identifiers | grep batmon`<br/>result:<br/>`- addon_xxxxxxxx_batmon`
  * use the identifier to limit the log output to the batmon addon and display the last 1000 lines:<br/>`ha host logs --identifier addon_xxxxxxxx_batmon -n 1000`<br/>where `xxxxxxxx` obviously should be the value returned by the previous command
  * you can also use the `--follow` option described at  `ha host logs --help`

### Error Log
Start of log:
```
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
16:41:36 INFO [bt] BT Discovery:
16:41:41 INFO [bt] BT XX:XX:XX:XX:XX:XX              SX100-B230019
16:41:41 INFO [main] Bleak version 0.13.1a1, BtBackend version bluez-v5.70
16:41:41 INFO [main] connecting mqtt batmon2ha@core-mosquitto
/app/main.py:196: DeprecationWarning: Callback API version 1 is deprecated, update to latest version
  mqtt_client = paho.Client(paho.CallbackAPIVersion.VERSION1)                                                                        16:41:41 INFO [main] Fetching 1 BMS + 0 virtual + 0 others serially, period=1.00s, keep_alive=True                                   16:41:41 INFO [sampling] connecting bms SuperVoltBt(XX:XX:XX:XX:XX:XX,battery_sx100)                                                 16:41:41 INFO [main] mqtt watchdog loop started with timeout 300.0s                                                                  16:41:43 ERROR [sampling] SuperVoltBt(XX:XX:XX:XX:XX:XX,battery_sx100) device not found, retry in 1 seconds                          16:41:43 ERROR [supervolt] (<class 'bleak.exc.BleakError'>, BleakError('Characteristic 6e400002-b5a3-f393-e0a9-e50e24dcca9e was not fTraceback (most recent call last):
  File "/app/bmslib/models/supervolt.py", line 113, in requestData
    await self.requestRealtimeData()
  File "/app/bmslib/models/supervolt.py", line 95, in requestRealtimeData
    ret = await self.client.write_gatt_char(char_specifier=handle, data=data)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/venv/lib/python3.11/site-packages/bleak/backends/bluezdbus/client.py", line 904, in write_gatt_char
[... 85 lines cut ...]
    raise exceptions[0]
  File "/app/main.py", line 314, in fn
    await t()
  File "/app/bmslib/sampling.py", line 187, in __call__
    await bms.disconnect()
  File "/app/bmslib/models/supervolt.py", line 84, in disconnect
    await self.client.stop_notify(self.UUID_RX)
  File "/app/venv/lib/python3.11/site-packages/bleak/backends/bluezdbus/client.py", line 1099, in stop_notify
    raise BleakError("Characteristic {} not found!".format(char_specifier))
bleak.exc.BleakError: Characteristic 6e400003-b5a3-f393-e0a9-e50e24dcca9e not found!

16:45:05 WARNING [main] too many errors, abort
16:45:05 INFO [main] fetch_loop <function main.<locals>.fn at 0x7faf4d6020> ends
16:45:05 INFO [main] All fetch loops ended. shutdown is already False
16:45:05 INFO [main] Disconnecting SuperVoltBt(XX:XX:XX:XX:XX:XX,battery_sx100)
16:45:05 INFO [main] exit signal handler... (), {}, shutdown was True
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped
```

fl4p, 2024-05-16:
Hi,
apparently they changed the BLE characteristic UID. Someone needs with this BMS needs to run [service_explorer.py](https://github.com/fl4p/batmon-ha/blob/master/tools/service_explorer.py)

nightcat91, 2024-05-16:
I got these documents from Supervolt in March 2024 via an email contact.
Maybe they are useful?

At least in the bt-protocol PDF there are some UUIDs on the last page and a lot more info about the full BT protocol in general.

bt_protocol.pdf
https://support.supervolt.de/attachments/token/VsJUkKGbW5zPzOdH5dM5MZbqt/?name=%5BB%5D+bt+protocol.pdf

rs485_protocol.docx
https://support.supervolt.de/attachments/token/rCajBS0UYSQo4cwcP0ugLNWxF/?name=%5BB%5D+rs485+protocol.docx

tinkercasa, 2024-05-16:
Executing  [service_explorer.py](https://github.com/fl4p/batmon-ha/blob/master/tools/service_explorer.py) confirmed that the UUIDs at the end of the PDF [[B] bt protocol.pdf](https://support.supervolt.de/attachments/token/VsJUkKGbW5zPzOdH5dM5MZbqt/?name=%5BB%5D+bt+protocol.pdf) are correct.

```
$ python3 service_explorer.py
INFO:__main__:Connecting XX:XX:XX:XX:XX:XX
INFO:__main__:Connected: True
INFO:__main__:[Service] 00002600-0000-1000-8000-00805f9b34fb (Handle: 23): Vendor specific
INFO:__main__:  [Characteristic] 00007000-0000-1000-8000-00805f9b34fb (Handle: 24): Vendor specific (write,indicate), Value: None
INFO:__main__:          [Descriptor] 00002902-0000-1000-8000-00805f9b34fb (Handle: 26): Client Characteristic Configuration) | Value: b'\x00\x00'
INFO:__main__:  [Characteristic] 00007001-0000-1000-8000-00805f9b34fb (Handle: 27): Vendor specific (read,write-without-response), Value: b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'
INFO:__main__:[Service] 0000ff00-0000-1000-8000-00805f9b34fb (Handle: 14): Vendor specific
INFO:__main__:  [Characteristic] 0000ff01-0000-1000-8000-00805f9b34fb (Handle: 17): Vendor specific (notify), Value: None
INFO:__main__:          [Descriptor] 00002902-0000-1000-8000-00805f9b34fb (Handle: 19): Client Characteristic Configuration) | Value: b'\x01\x00'
INFO:__main__:  [Characteristic] 0000ff02-0000-1000-8000-00805f9b34fb (Handle: 15): Vendor specific (write-without-response), Value: None
INFO:__main__:[Service] 0000180a-0000-1000-8000-00805f9b34fb (Handle: 20): Device Information
INFO:__main__:  [Characteristic] 00002a23-0000-1000-8000-00805f9b34fb (Handle: 21): System ID (read), Value: b'\xc7AP,\x0e\xe3'
INFO:__main__:[Service] 00001801-0000-1000-8000-00805f9b34fb (Handle: 6): Generic Attribute Profile
INFO:__main__:  [Characteristic] 00002b2a-0000-1000-8000-00805f9b34fb (Handle: 12): Database Hash (read), Value: b'\xd2\xa3w\x8d\xd7\n\xf5T\x91\xd9d\xc88i\xd1\x8b'
INFO:__main__:  [Characteristic] 00002a05-0000-1000-8000-00805f9b34fb (Handle: 7): Service Changed (indicate), Value: None
INFO:__main__:          [Descriptor] 00002902-0000-1000-8000-00805f9b34fb (Handle: 9): Client Characteristic Configuration) | Value: b'\x02\x00'
INFO:__main__:  [Characteristic] 00002b29-0000-1000-8000-00805f9b34fb (Handle: 10): Client Supported Features (read,write), Value: b'\x01'
```

nightcat91, 2024-06-02:
@fl4p 
Will it be possible to fix the Supervolt integration with the new UUIDs?

I would love to test it once available. Thanks

koebiii, 2024-06-11:
So I guess supervolt.py would have to be modified in line 19 and 20 to hopefully get the newer (v3?) BMS versions connected... likely breaking connection to previous versions.

Today:

19 UUID_RX = '6e400003-b5a3-f393-e0a9-e50e24dcca9e' # std uart TX (tx on device side, rx on host)
20 UUID_TX = '6e400002-b5a3-f393-e0a9-e50e24dcca9e' # std uart RX

bt_protocol.pdf states:

Bluetooth UUID
SERVICE UUID: 0000ff00-0000-1000-8000-00805f9b34fb
write characteristic UUID: 0000ff02-0000-1000-8000-00805f9b34fb
read characteristic UUID: 0000ff01-0000-1000-8000-00805f9b34fb

Requested:

19 UUID_RX = '0000ff01-0000-1000-8000-00805f9b34fb' # std uart TX (tx on device side, rx on host)
20 UUID_TX = '0000ff02-0000-1000-8000-00805f9b34fb' # std uart RX

nightcat91, 2024-06-14:
Is there somehow a way for us to do this change just locally with the already installed version?

I wouldnt mind changing the file and testing it, but I dont have the knowledge to make a pull request on github or clone the repository and be able to install another version in my HomeAssistant...

Otherwise I think only @fl4p can make that change and push a new version.

koebiii, 2024-06-14:
You can run the stand-alone setup, see https://github.com/fl4p/batmon-ha/blob/master/doc/Standalone.md. I was able to set it up and edit supervolt.py on Ubuntu. I ran out of time when I came to installing a local mqtt instance. This will have to wait for a few more days unfortunately...

nightcat91, 2024-06-14:
This is running as a HomeAssistant addon in my camper. I can´t run another standalone linux there sadly.
Thats why I asked about beeing able to change that one file locally inside the installed HomeAssistant Addon?

Karatzie, 2024-06-20:
Hi.

please do not change the old Supervolt code.
Please add a new type Supervolt 2024 Edition in a seperate code.

Thanks

BR 
Andreas

smartloh, 2024-08-20:
Hello,
I also have a Supervolt from 2024, which writes the error below in the log. Regarding the installation, I have the Batmon Version of @nightcat91. In this version line 19/20 of supervolt.py is already fixed.

It would be really great, if someone can let me know, how i can read out the Supervolt battery.

Thanks for your help,
Joerg

address: CB:04:0E:7E:4D:88
  type: supervolt
  alias: camper

14:24:47 INFO [sampling] connecting bms SuperVoltBt(CB:04:0E:7E:4D:88,camper)
14:24:48 INFO [sampling] connected bms SuperVoltBt(CB:04:0E:7E:4D:88,camper)!
14:25:08 ERROR [sampling] camper error (#7): '>' not supported between instances of 'NoneType' and 'int'
Traceback (most recent call last):
  File "/app/bmslib/sampling.py", line 155, in __call__
    s = await self._sample_inner()
        ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/bmslib/sampling.py", line 237, in _sample_inner
    sample = await bms.fetch()
             ^^^^^^^^^^^^^^^^^
  File "/app/bmslib/models/supervolt.py", line 377, in fetch
    sample = BmsSample(
             ^^^^^^^^^^
  File "/app/bmslib/bms.py", line 64, in __init__
    if capacity > 0 and (math.isnan(soc) or (isinstance(soc, int) and charge > 0)):
       ^^^^^^^^^^^^
TypeError: '>' not supported between instances of 'NoneType' and 'int'
14:25:08 INFO [sampling] Bleak version 0.13.1a1
14:25:08 ERROR [main] 1 exceptions occurred fetching BMSs
14:25:08 ERROR [main] Error (num 6, max 200) reading BMS: '>' not supported between instances of 'NoneType' and 'int'
14:25:08 ERROR [main] Stack: Traceback (most recent call last):
  File "/app/main.py", line 36, in fetch_loop
    if await fn():
       ^^^^^^^^^^
  File "/app/main.py", line 319, in fn
    raise exceptions[0]
  File "/app/main.py", line 314, in fn
    await t()
  File "/app/bmslib/sampling.py", line 155, in __call__
    s = await self._sample_inner()
        ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/bmslib/sampling.py", line 237, in _sample_inner
    sample = await bms.fetch()
             ^^^^^^^^^^^^^^^^^
  File "/app/bmslib/models/supervolt.py", line 377, in fetch
    sample = BmsSample(
             ^^^^^^^^^^
  File "/app/bmslib/bms.py", line 64, in __init__
    if capacity > 0 and (math.isnan(soc) or (isinstance(soc, int) and charge > 0)):
       ^^^^^^^^^^^^
TypeError: '>' not supported between instances of 'NoneType' and 'int'

14:25:30 ERROR [sampling] camper error (#8): '>' not supported between instances of 'NoneType' and 'int'
Traceback (most recent call last):
  File "/app/bmslib/sampling.py", line 155, in __call__
    s = await self._sample_inner()
        ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/bmslib/sampling.py", line 237, in _sample_inner
    sample = await bms.fetch()
             ^^^^^^^^^^^^^^^^^
  File "/app/bmslib/models/supervolt.py", line 377, in fetch
    sample = BmsSample(
             ^^^^^^^^^^
  File "/app/bmslib/bms.py", line 64, in __init__
    if capacity > 0 and (math.isnan(soc) or (isinstance(soc, int) and charge > 0)):
       ^^^^^^^^^^^^
TypeError: '>' not supported between instances of 'NoneType' and 'int'
14:25:30 INFO [sampling] Bleak version 0.13.1a1
14:25:30 WARNING [sampling] camper disconnect because no data has been flowing for some time
14:25:33 WARNING [bt] BMS SuperVoltBt(CB:04:0E:7E:4D:88,camper) disconnected after 44.9s!
14:25:33 ERROR [main] 1 exceptions occurred fetching BMSs
14:25:33 ERROR [main] Error (num 7, max 200) reading BMS: '>' not supported between instances of 'NoneType' and 'int'
14:25:33 ERROR [main] Stack: Traceback (most recent call last):
  File "/app/main.py", line 36, in fetch_loop
    if await fn():
       ^^^^^^^^^^
  File "/app/main.py", line 319, in fn
    raise exceptions[0]
  File "/app/main.py", line 314, in fn
    await t()
  File "/app/bmslib/sampling.py", line 155, in __call__
    s = await self._sample_inner()
        ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/bmslib/sampling.py", line 237, in _sample_inner
    sample = await bms.fetch()
             ^^^^^^^^^^^^^^^^^
  File "/app/bmslib/models/supervolt.py", line 377, in fetch
    sample = BmsSample(
             ^^^^^^^^^^
  File "/app/bmslib/bms.py", line 64, in __init__
    if capacity > 0 and (math.isnan(soc) or (isinstance(soc, int) and charge > 0)):
       ^^^^^^^^^^^^
TypeError: '>' not supported between instances of 'NoneType' and 'int'

14:25:35 INFO [sampling] connecting bms SuperVoltBt(CB:04:0E:7E:4D:88,camper)
14:25:36 INFO [sampling] connected bms SuperVoltBt(CB:04:0E:7E:4D:88,camper)!

fl4p, 2025-02-02:
Hi,
I've add the new char UUIDS : https://github.com/fl4p/batmon-ha/commit/299a85702a2e36bd9f91d69860a20fea17e9f6af

can someone confirm that it works?

tinkercasa, 2025-02-05:
Hi fl4p,

thank you for addressing the issues with the newer supervolt bms.
Here is my log output of batmon v1.82:

`s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
21:30:52 INFO [main] Started in pair-only mode (bleak 0.20.2)
21:30:52 INFO [main] No PSK, nothing to pair
21:30:52 INFO [main] exit signal handler... (), {}, shutdown was False
21:30:53 INFO [bt] BT Discovery:
21:30:58 INFO [bt] BT A0:B0:C0:D0:E0:F0              SX100-B000000
21:30:58 INFO [main] Bleak version 0.20.2, BtBackend version bluez-v5.79
21:30:58 INFO [main] connecting mqtt batmon2ha@core-mosquitto
21:30:58 INFO [main] Fetching 1 BMS + 0 virtual + 0 others serially, period=1.00s, keep_alive=True
21:30:58 INFO [sampling] connecting bms SuperVoltBt(A0:B0:C0:D0:E0:F0,battery_sx100)
21:30:58 INFO [main] mqtt watchdog loop started with timeout 300.0s
21:31:00 INFO [supervolt] tried char rx/tx='6e400003-b5a3-f393-e0a9-e50e24dcca9e'/'6e400002-b5a3-f393-e0a9-e50e24dcca9e', err Characteristic 6e400002-b5a3-f393-e0a9-e50e24dcca9e was not found!
21:31:00 INFO [supervolt] tried char rx/tx='0000ff02-0000-1000-8000-00805f9b34fb'/'0000ff01-0000-1000-8000-00805f9b34fb', err Characteristic 0000ff01-0000-1000-8000-00805f9b34fb does not support write operations!
21:31:00 INFO [bt] [Service] 00002600-0000-1000-8000-00805f9b34fb (Handle: 23): Vendor specific
21:31:00 INFO [bt] 	[Characteristic] 00007001-0000-1000-8000-00805f9b34fb (Handle: 27): Vendor specific (read,write-without-response), Value: b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'
21:31:00 INFO [bt] 	[Characteristic] 00007000-0000-1000-8000-00805f9b34fb (Handle: 24): Vendor specific (write,indicate), Value: None
21:31:00 INFO [bt] 		[Descriptor] 00002902-0000-1000-8000-00805f9b34fb (Handle: 26): Client Characteristic Configuration) | Value: b'\x00\x00'
21:31:00 INFO [bt] [Service] 0000ff00-0000-1000-8000-00805f9b34fb (Handle: 14): Vendor specific
21:31:00 INFO [bt] 	[Characteristic] 0000ff01-0000-1000-8000-00805f9b34fb (Handle: 17): Vendor specific (notify), Value: None
21:31:00 INFO [bt] 		[Descriptor] 00002902-0000-1000-8000-00805f9b34fb (Handle: 19): Client Characteristic Configuration) | Value: b'\x01\x00'
21:31:00 INFO [bt] 	[Characteristic] 0000ff02-0000-1000-8000-00805f9b34fb (Handle: 15): Vendor specific (write-without-response), Value: None
21:31:00 INFO [bt] [Service] 0000180a-0000-1000-8000-00805f9b34fb (Handle: 20): Device Information
21:31:00 INFO [bt] 	[Characteristic] 00002a23-0000-1000-8000-00805f9b34fb (Handle: 21): System ID (read), Value: b'\xc7AP,\x0e\xe3'
21:31:00 INFO [bt] [Service] 00001801-0000-1000-8000-00805f9b34fb (Handle: 6): Generic Attribute Profile
21:31:00 INFO [bt] 	[Characteristic] 00002b29-0000-1000-8000-00805f9b34fb (Handle: 10): Client Supported Features (read,write), Value: b'\x01'
21:31:00 INFO [bt] 	[Characteristic] 00002a05-0000-1000-8000-00805f9b34fb (Handle: 7): Service Changed (indicate), Value: None
21:31:00 INFO [bt] 		[Descriptor] 00002902-0000-1000-8000-00805f9b34fb (Handle: 9): Client Characteristic Configuration) | Value: b'\x02\x00'
21:31:00 INFO [bt] 	[Characteristic] 00002b2a-0000-1000-8000-00805f9b34fb (Handle: 12): Database Hash (read), Value: b'\xd2\xa3w\x8d\xd7\n\xf5T\x91\xd9d\xc88i\xd1\x8b'
21:31:00 ERROR [sampling] battery_sx100 error (#1): Notify characteristic (rx) not found
Traceback (most recent call last):
  File "/app/bmslib/sampling.py", line 155, in __call__
    s = await self._sample_inner()
        ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/bmslib/sampling.py", line 229, in _sample_inner
    async with bms:
               ^^^
  File "/app/bmslib/bt.py", line 348, in __aenter__
    await self.connect()
  File "/app/bmslib/models/supervolt.py", line 101, in connect
    raise Exception("Notify characteristic (rx) not found")
Exception: Notify characteristic (rx) not found
21:31:00 INFO [sampling] Bleak version 0.20.2
21:31:00 ERROR [supervolt] (<class 'bleak.exc.BleakError'>, BleakError('Characteristic  was not found!'), <traceback object at 0x7f837dc5c0>)
Traceback (most recent call last):
  File "/app/bmslib/models/supervolt.py", line 137, in requestData
    await self.requestRealtimeData()
  File "/app/bmslib/models/supervolt.py", line 119, in requestRealtimeData
    ret = await self.client.write_gatt_char(char_specifier=handle, data=data)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/venv/lib/python3.12/site-packages/bleak/__init__.py", line 659, in write_gatt_char
    await self._backend.write_gatt_char(char_specifier, data, response)
  File "/app/venv/lib/python3.12/site-packages/bleak/backends/bluezdbus/client.py", line 848, in write_gatt_char
    raise BleakError("Characteristic {0} was not found!".format(char_specifier))
bleak.exc.BleakError: Characteristic  was not found!
21:31:00 ERROR [sampling] battery_sx100 error (#2): '>' not supported between instances of 'NoneType' and 'int'
Traceback (most recent call last):
  File "/app/bmslib/sampling.py", line 155, in __call__
    s = await self._sample_inner()
[... 10 lines cut ...]
[... cut ...]
    s = await self._sample_inner()
        ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/bmslib/sampling.py", line 239, in _sample_inner
    sample = await bms.fetch()
             ^^^^^^^^^^^^^^^^^
  File "/app/bmslib/models/supervolt.py", line 407, in fetch
    sample = BmsSample(
             ^^^^^^^^^^
  File "/app/bmslib/bms.py", line 64, in __init__
    if capacity > 0 and (math.isnan(soc) or (isinstance(soc, int) and charge > 0)):
       ^^^^^^^^^^^^
TypeError: '>' not supported between instances of 'NoneType' and 'int'`

fl4p, 2025-02-06:
@tinkercasa thanks for posting the logs, it looks like the notify/write (rx/tx) specifiers are just swapped. I'll fix this with the next update.

> I guess _supervolt.py_ would have to be modified to hopefully get the newer (v3) BMS versions connected... likely breaking connection to previous (v2?) BMS version(s).
> 
> _supervolt.py_ in line 19 and 20 today:
> 
> ```
> UUID_RX = '6e400003-b5a3-f393-e0a9-e50e24dcca9e' # std uart TX (tx on device side, rx on host)
> UUID_TX = '6e400002-b5a3-f393-e0a9-e50e24dcca9e' # std uart RX
> ```
> 
> bt_protocol.pdf states:
> 
> SERVICE UUID: 0000ff00-0000-1000-8000-00805f9b34fb write characteristic UUID: 0000ff02-0000-1000-8000-00805f9b34fb read characteristic UUID: 0000ff01-0000-1000-8000-00805f9b34fb
> 
> _supervolt.py_ change request:
> 
> ```
> UUID_RX = '0000ff02-0000-1000-8000-00805f9b34fb' # std uart TX (tx on device side, rx on host)
> UUID_TX = '0000ff01-0000-1000-8000-00805f9b34fb' # std uart RX
> ```

I think the changed `UUID_RX` and `UUID_TX` need to be swapped. UART RX/TX is naming is quite confusing since its about the perspective of the device

tinkercasa, 2025-02-06:
As Karatzie/Andreas pointed out above that it might be advisable to separate the new supervolt BMS settings from the old ones in a differnet BMS file (like supervolt_v3.py). 
I checked the supervolt webpage. The latest version of the supervolt app is capable of updating the BMS version of the connceted battery if a newer BMS version is available.
I don't own an older supervolt battery model so I can't check if it's possible to update older batteries to the newest BMS. If that was possible there would be no need to support different supervolt BMS versions in batmon.

fl4p, 2025-02-06:
I don't see a reason for adding a separate BMS type such as `supervolt_v3`. It is easy to detect which version batmon is connectiog to, as it just need to check what BLE Characteristics the device has. It already does it with Daly BMS

tinkercasa, 2025-02-12:
I didn't know you were already differentiating varying BMS versions in one vendor specific file. That's great.

I contacted the Supervolt service about the different battery and BMS versions. Older batteries can only be updated via a data cable that can be supplied by the service. 
This information is kind of obselete now that I know you will support old and new BMS versions anyway.