fl4p/batmon-ha #227: JK_B1A20S15P cannot read or publish data?

ligius-, 2024-05-03:
I have a strange issue with my JK BMS above. I can see that some values are sampled in the logs but they never appear in MQTT, I assume due to the exception below. It's configured without a PIN as with a PIN it times out.

Short sample:
```
May 03 16:12:13 beepi c2395a33a9ff[1013]: 17:12:13 INFO [sampling] battery1 Power jump 334 % (prev=0 last=401, REG=120)
May 03 16:12:13 beepi c2395a33a9ff[1013]: 17:12:13 INFO [sampling] battery1: BmsSampl(47.5%,U=53.1V,I=7.55A,P=401W,Q=67/140Ah,mos=nan°C)
May 03 16:12:13 beepi c2395a33a9ff[1013]: 17:12:13 INFO [sampling] battery1 volt=[3319,3319,3317,3321,3321,3321,3324,3321,3321,3319,3320,3320,3321,3321,3322,3320] temp=[nan, nan, nan, nan]
May 03 16:12:13 beepi c2395a33a9ff[1013]: 17:12:13 INFO [sampling] Sending HA discovery for battery1 (num_samples=0)
```

This is running under Home Assistant in Docker under rPi 4, the internal MQTT broker is configured and is already used for other topics. I also tried "concurrent sampling" with and without disabling the keepalive.

YAML:
```
devices:
  - address: C8:47:80:03:7F:78
    type: jk
    alias: battery1
    debug: true
    current_calibration: 1
concurrent_sampling: false
keep_alive: true
verbose_log: true
sample_period: 3
publish_period: 2
invert_current: true
watchdog: true
expire_values_after: 20
install_newer_bleak: true
bt_power_cycle: true
```

LOGS:
```
May 03 16:12:13 beepi c2395a33a9ff[1013]: 17:12:13 DEBUG [manager] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_C8_47_80_03_7F_78/service000e/char0011): ['org.bluez.GattCharacteristic1', {'Value': <dbus_fast.signature.Variant ('ay', bytearray(b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xef\x00\x00\x00\x00\x00'))>}, []]
May 03 16:12:13 beepi c2395a33a9ff[1013]: 17:12:13 DEBUG [jikong] bms msg(22) (buf150): 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 ef 0 0 0 0 0
May 03 16:12:13 beepi c2395a33a9ff[1013]:
May 03 16:12:13 beepi c2395a33a9ff[1013]: 17:12:13 DEBUG [manager] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_C8_47_80_03_7F_78/service000e/char0011): ['org.bluez.GattCharacteristic1', {'Value': <dbus_fast.signature.Variant ('ay', bytearray(b'\x7f\xcf\x00\x00\x98\x1e\x06\x00~\x1d\x00\x00\x01\x01\xfd\x00\x00\x00\x08\x00\x00\x00\x000\xd9\x03\x01\x00\xe0"\x02\x00\x1f\x00\x00\x00N\x1aC\x00d\x00\x00\x00\xc6\x03\xc4\x00\x01\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xff\x00\x01\x00\x00\x00\xc8\x03\'\x00\x00\x00p\x82A@\x00\x00\x00\x00\xbf\x14\xa1\x1d\x00\x01\x00\x01\xb6\x05\x00\x00\x1d\xdc\xd5\x02\x00\x00\x00\x00\xef\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'))>}, []]
May 03 16:12:13 beepi c2395a33a9ff[1013]: 17:12:13 DEBUG [jikong] bms msg(128) (buf278): 7f cf 0 0 98 1e 6 0 7e 1d 0 0 1 1 fd 0 0 0 8 0 0 0 0 30 d9 3 1 0 e0 22 2 0 1f 0 0 0 4e 1a 43 0 64 0 0 0 c6 3 c4 0 1 1 0 0 0 0 0 0 0 0 0 0 0 0 0 0 ff 0 1 0 0 0 c8 3 27 0 0 0 70 82 41 40 0 0 0 0 bf 14 a1 1d 0 1 0 1 b6 5 0 0 1d dc d5 2 0 0 0 0 ef 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0
May 03 16:12:13 beepi c2395a33a9ff[1013]:
May 03 16:12:13 beepi c2395a33a9ff[1013]: 17:12:13 DEBUG [manager] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_C8_47_80_03_7F_78/service000e/char0011): ['org.bluez.GattCharacteristic1', {'Value': <dbus_fast.signature.Variant ('ay', bytearray(b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\xfe\xff\x7f\xdc\x0f\x01\x00\x00\x00\x00\x00\x00\xbe'))>}, []]
May 03 16:12:13 beepi c2395a33a9ff[1013]: 17:12:13 DEBUG [jikong] bms msg(22) (buf300): 0 0 0 0 0 0 0 0 0 fe ff 7f dc f 1 0 0 0 0 0 0 be
May 03 16:12:13 beepi c2395a33a9ff[1013]:
May 03 16:12:13 beepi c2395a33a9ff[1013]: 17:12:13 DEBUG [jikong] got response 2 (len300)
May 03 16:12:13 beepi c2395a33a9ff[1013]: 17:12:13 DEBUG [jikong] New 11.x firmware, offset=32
May 03 16:12:13 beepi c2395a33a9ff[1013]: 17:12:13 INFO [sampling] battery1 Power jump 334 % (prev=0 last=401, REG=120)
May 03 16:12:13 beepi c2395a33a9ff[1013]: 17:12:13 INFO [sampling] battery1: BmsSampl(47.5%,U=53.1V,I=7.55A,P=401W,Q=67/140Ah,mos=nan°C)
May 03 16:12:13 beepi c2395a33a9ff[1013]: 17:12:13 INFO [sampling] battery1 volt=[3319,3319,3317,3321,3321,3321,3324,3321,3321,3319,3320,3320,3321,3321,3322,3320] temp=[nan, nan, nan, nan]
May 03 16:12:13 beepi c2395a33a9ff[1013]: 17:12:13 INFO [sampling] Sending HA discovery for battery1 (num_samples=0)
May 03 16:12:13 beepi c2395a33a9ff[1013]: 17:12:13 DEBUG [mqtt_util] discovery msg homeassistant/sensor/battery1/_soc_total_voltage/config: {"unique_id": "battery1__soc_total_voltage", "name": "voltage", "device_class": "voltage", "state_class": "measurement", "unit_of_measurement": "V", "state_topic": "battery1/soc/total_voltage", "expire_after": 20.0, "device": {"identifiers": ["3080342450"], "manufacturer": "JK", "name": "JK_B1A20S15P (battery1)", "model": "JK_B1A20S15P", "sw_version": "11.288", "hw_version": "11.XW"}, "icon": "mdi:meter-electric"}
May 03 16:12:13 beepi c2395a33a9ff[1013]: 17:12:13 DEBUG [mqtt_util] discovery msg homeassistant/sensor/battery1/_soc_current/config: {"unique_id": "battery1__soc_current", "name": "current", "device_class": "current", "state_class": "measurement", "unit_of_measurement": "A", "state_topic": "battery1/soc/current", "expire_after": 20.0, "device": {"identifiers": ["3080342450"], "manufacturer": "JK", "name": "JK_B1A20S15P (battery1)", "model": "JK_B1A20S15P", "sw_version": "11.288", "hw_version": "11.XW"}}
May 03 16:12:13 beepi c2395a33a9ff[1013]: 17:12:13 DEBUG [mqtt_util] discovery msg homeassistant/sensor/battery1/_soc_balance_current/config: {"unique_id": "battery1__soc_balance_current", "name": "balance_current", "device_class": "current", "state_class": "measurement", "unit_of_measurement": "A", "state_topic": "battery1/soc/balance_current", "expire_after": 20.0, "device": {"identifiers": ["3080342450"], "manufacturer": "JK", "name": "JK_B1A20S15P (battery1)", "model": "JK_B1A20S15P", "sw_version": "11.288", "hw_version": "11.XW"}, "icon": "mdi:scale-unbalanced"}
May 03 16:12:13 beepi c2395a33a9ff[1013]: 17:12:13 DEBUG [mqtt_util] discovery msg homeassistant/sensor/battery1/_soc_soc_percent/config: {"unique_id": "battery1__soc_soc_percent", "name": "soc", "device_class": "battery", "unit_of_measurement": "%", "state_topic": "battery1/soc/soc_percent", "expire_after": 20.0, "device": {"identifiers": ["3080342450"], "manufacturer": "JK", "name": "JK_B1A20S15P (battery1)", "model": "JK_B1A20S15P", "sw_version": "11.288", "hw_version": "11.XW"}, "icon": "mdi:battery"}
May 03 16:12:13 beepi c2395a33a9ff[1013]: 17:12:13 DEBUG [mqtt_util] discovery msg homeassistant/sensor/battery1/_soc_power/config: {"unique_id": "battery1__soc_power", "name": "power", "device_class": "power", "state_class": "measurement", "unit_of_measurement": "W", "state_topic": "battery1/soc/power", "expire_after": 20.0, "device": {"identifiers": ["3080342450"], "manufacturer": "JK", "name": "JK_B1A20S15P (battery1)", "model": "JK_B1A20S15P", "sw_version": "11.288", "hw_version": "11.XW"}, "icon": "mdi:flash"}
May 03 16:12:13 beepi c2395a33a9ff[1013]: 17:12:13 DEBUG [mqtt_util] discovery msg homeassistant/sensor/battery1/_soc_capacity/config: {"unique_id": "battery1__soc_capacity", "name": "capacity", "unit_of_measurement": "Ah", "state_topic": "battery1/soc/capacity", "expire_after": 20.0, "device": {"identifiers": ["3080342450"], "manufacturer": "JK", "name": "JK_B1A20S15P (battery1)", "model": "JK_B1A20S15P", "sw_version": "11.288", "hw_version": "11.XW"}}
May 03 16:12:13 beepi c2395a33a9ff[1013]: 17:12:13 DEBUG [mqtt_util] discovery msg homeassistant/sensor/battery1/_soc_cycle_capacity/config: {"unique_id": "battery1__soc_cycle_capacity", "name": "cycle_capacity", "unit_of_measurement": "Ah", "state_topic": "battery1/soc/cycle_capacity", "expire_after": 20.0, "device": {"identifiers": ["3080342450"], "manufacturer": "JK", "name": "JK_B1A20S15P (battery1)", "model": "JK_B1A20S15P", "sw_version": "11.288", "hw_version": "11.XW"}}
May 03 16:12:13 beepi c2395a33a9ff[1013]: 17:12:13 DEBUG [mqtt_util] discovery msg homeassistant/sensor/battery1/_soc_num_cycles/config: {"unique_id": "battery1__soc_num_cycles", "name": "num_cycles", "state_class": "measurement", "unit_of_measurement": "N", "state_topic": "battery1/soc/num_cycles", "expire_after": 20.0, "device": {"identifiers": ["3080342450"], "manufacturer": "JK", "name": "JK_B1A20S15P (battery1)" [... cut ...]
May 03 16:12:21 beepi c2395a33a9ff[1013]: 17:12:21 DEBUG [manager] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_C8_47_80_03_7F_78/service000e/char0011): ['org.bluez.GattCharacteristic1', {'Notifying': <dbus_fast.signature.Variant ('b', False)>}, []]
May 03 16:12:24 beepi c2395a33a9ff[1013]: 17:12:24 INFO [sampling] connecting bms JKBt(C8:47:80:03:7F:78,battery1)
May 03 16:12:24 beepi c2395a33a9ff[1013]: 17:12:24 INFO [bt] connecting battery1 (C8:47:80:03:7F:78) adapter=default timeout=6
May 03 16:12:24 beepi c2395a33a9ff[1013]: 17:12:24 DEBUG [client] Connecting to device @ C8:47:80:03:7F:78
```

ligius-, 2024-05-03:
I left it running with a delay for sampling of 5s and it does manage to publish values about 1% of the time. Here is a successful snippet:

```
May 03 17:15:14 beepi f4e90f3202a0[1013]:
May 03 17:15:15 beepi f4e90f3202a0[1013]: 18:15:15 DEBUG [manager] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_C8_47_80_03_7F_78/service000e/char0011): ['org.bluez.GattCharacteristic1', {'Value': <dbus_fast.signature.Variant ('ay', bytearray(b'\x00\x00\xe0"\x02\x00\xdc\x05\x00\x00\\\r\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'))>}, []]
May 03 17:15:15 beepi f4e90f3202a0[1013]: 18:15:15 DEBUG [jikong] bms msg(22) (buf150): 0 0 e0 22 2 0 dc 5 0 0 5c d 0 0 0 0 0 0 0 0 0 0
May 03 17:15:15 beepi f4e90f3202a0[1013]:
May 03 17:15:16 beepi f4e90f3202a0[1013]: 18:15:16 DEBUG [manager] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_C8_47_80_03_7F_78/service000e/char0011): ['org.bluez.GattCharacteristic1', {'Value': <dbus_fast.signature.Variant ('ay', bytearray(b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'))>}, []]
May 03 17:15:16 beepi f4e90f3202a0[1013]: 18:15:16 DEBUG [jikong] bms msg(128) (buf278): 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0
May 03 17:15:16 beepi f4e90f3202a0[1013]:
May 03 17:15:16 beepi f4e90f3202a0[1013]: 18:15:16 DEBUG [manager] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_C8_47_80_03_7F_78/service000e/char0011): ['org.bluez.GattCharacteristic1', {'Value': <dbus_fast.signature.Variant ('ay', bytearray(b'\xb0q\x0b\x00\x08\x00<2\x00\xfc\xf9\xff\xff\x1f-\x00\x00\x00\x00\x00\x00\xb3'))>}, []]
May 03 17:15:16 beepi f4e90f3202a0[1013]: 18:15:16 DEBUG [jikong] bms msg(22) (buf300): b0 71 b 0 8 0 3c 32 0 fc f9 ff ff 1f 2d 0 0 0 0 0 0 b3
May 03 17:15:16 beepi f4e90f3202a0[1013]:
May 03 17:15:16 beepi f4e90f3202a0[1013]: 18:15:16 DEBUG [jikong] got response 1 (len300)
May 03 17:15:16 beepi f4e90f3202a0[1013]: 18:15:16 DEBUG [manager] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_C8_47_80_03_7F_78/service000e/char0011): ['org.bluez.GattCharacteristic1', {'Value': <dbus_fast.signature.Variant ('ay', bytearray(b'\xaaU\x90\xeb\xc8\x01\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00D'))>}, []]
May 03 17:15:16 beepi f4e90f3202a0[1013]: 18:15:16 DEBUG [jikong] bms msg(20) (buf20): aa 55 90 eb c8 1 1 0 0 0 0 0 0 0 0 0 0 0 0 44
May 03 17:15:16 beepi f4e90f3202a0[1013]:
May 03 17:15:16 beepi f4e90f3202a0[1013]: 18:15:16 DEBUG [manager] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_C8_47_80_03_7F_78/service000e/char0011): ['org.bluez.GattCharacteristic1', {'Value': <dbus_fast.signature.Variant ('ay', bytearray(b"U\xaa\xeb\x90\x02\xb1\xd9\x0c\xda\x0c\xd8\x0c\xd8\x0c\xca\x0c\xd6\x0c\xd9\x0c\xc6\x0c\xcc\x0c\xd8\x0c\xd1\x0c\xd6\x0c\xd6\x0c\xcc\x0c\xc6\x0c\xd3\x0c\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xff\xff\x00\x00\xd3\x0c\x15\x00\x00\x0e#\x00&\x00\'\x000\x002\x002\x002\x001\x001\x001\x002\x00 \x002\x001\x001\x002\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00"))>}, []]
May 03 17:15:16 beepi f4e90f3202a0[1013]: 18:15:16 DEBUG [jikong] header, clear buf bytearray(b'\xaaU\x90\xeb\xc8\x01\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00D')
May 03 17:15:16 beepi f4e90f3202a0[1013]: 18:15:16 DEBUG [jikong] bms msg(128) (buf128): 55 aa eb 90 2 b1 d9 c da c d8 c d8 c ca c d6 c d9 c c6 c cc c d8 c d1 c d6 c d6 c cc c c6 c d3 c 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 ff ff 0 0 d3 c 15 0 0 e 23 0 26 0 27 0 30 0 32 0 32 0 32 0 31 0 31 0 31 0 32 0 20 0 32 0 31 0 31 0 32 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0
May 03 17:15:16 beepi f4e90f3202a0[1013]:
May 03 17:15:17 beepi f4e90f3202a0[1013]: 18:15:17 DEBUG [manager] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_C8_47_80_03_7F_78/service000e/char0011): ['org.bluez.GattCharacteristic1', {'Value': <dbus_fast.signature.Variant ('ay', bytearray(b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xf3\x00\x00\x00\x00\x00'))>}, []]
May 03 17:15:17 beepi f4e90f3202a0[1013]: 18:15:17 DEBUG [jikong] bms msg(22) (buf150): 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 f3 0 0 0 0 0
May 03 17:15:17 beepi f4e90f3202a0[1013]:
May 03 17:15:18 beepi f4e90f3202a0[1013]: 18:15:18 DEBUG [manager] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_C8_47_80_03_7F_78/service000e/char0011): ['org.bluez.GattCharacteristic1', {'Value': <dbus_fast.signature.Variant ('ay', bytearray(b')\xcd\x00\x00\x7f\x8c\x13\x00\xb7\xa0\xff\xff\x04\x01\x00\x01\x00\x00\x08\x00\x00\x00\x003G\x17\x01\x00\xe [... cut ...] ss": "measurement", "unit_of_measurement": "V", "state_topic": "battery1/soc/total_voltage", "expire_after": 240.0, "device": {"identifiers": ["3080342450"], "manufacturer": "JK", "name": "JK_B1A20S15P (battery1)", "model": "JK_B1A20S15P", "sw_version": "11.288", "hw_version": "11.XW"}, "icon": "mdi:meter-electric"}
May 03 17:15:19 beepi f4e90f3202a0[1013]: 18:15:19 DEBUG [client] Sending PUBLISH (d0, q0, r0, m47), 'b'homeassistant/sensor/battery1/_soc_total_voltage/config'', ... (418 bytes)
```

Off-Grid-Garage, 2024-05-09:
Have you tried:
[image]

ligius-, 2024-05-09:
Yes, but that did not change the behavior. Strangely enough, it started working by itself correctly after a few hours. I suspect it expected a certain order of messages but the state was set differently by the BT app. That's just my assumption but no idea yet how the code actually works and what it expects.

ligius-, 2024-07-29:
Sadly, it still is spotty. Sometimes it has a good run, sometimes it fails to get any reading for 20+ minutes. Publish_period is 10s, sample_period is 5s, expiry is set to 600s.

[image]

The error is the same, I've seen it only a few times where it could not connect to the BMS but I assume that's something with the BT connectivity on the BMS side as I see this also in the app.