fl4p/batmon-ha #228: BMS Parameter

k2dp2k, 2024-05-14:
Hello,
are there any plans to make the settings of the jk inverter bms available ? eg. max cell voltage, balance voltage...
It would be a great thing if it is possible....

Off-Grid-Garage, 2024-05-22:
That is already working with the inverter BMS. The integration is just broken and I haven't seen the dev around for a while... not sure if this project is still supported or not.

[image]

k2dp2k, 2024-05-22:
Hi Andy :-) Sorry for getting me wrong.... I know that these settings are available to be read out, but i want them to be adjustable or settable through this plugin, like in the bluetooth app...

Off-Grid-Garage, 2024-05-22:
That won't work as all data is read only.

With the app you need to enter the BMS password first to get access to the
settings.

I don't think any of these projects here on GitHub can actually do that.

k2dp2k, 2024-05-22:
> That won't work as all data is read only. With the app you need to enter the BMS password first to get access to the settings. I don't think any of these projects here on GitHub can actually do that.
> […](#)

It is possible through esphome (https://github.com/syssi/esphome-jk-bms/issues/390), so i thought it could be possible here as well...

fl4p, 2025-08-30:
As @Off-Grid-Garage , batmon only reads values and controls switches.  There is no intention for it to be able to change BMS parameters.

closed 2025-08-30 by fl4p