fl4p/batmon-ha #229: Wrong values showed in relation to two Creabest VB024 batteries

Alex-DeIt, 2024-05-19:
Hello everybody,

I am able to read several values of two Creabest batteries using the "dummy" type setting in Batmon. Unfortunately the values imported to HA are very different from the values showed by the Creabest app and they are not plausible. Even the shown voltage of the two batteies is lower than it should be (both should have approx 13,6V) and it is going periodically up an down (see the immage?.

[image]

Does someone have a clue how to solve this problem?
Thank you in advance for any support you can give me.

snipah, 2024-05-20:
Are you sure, that dummy really catches any BT-data? I would assume, that it just creates some "dummy"-data and no really comms with the BMS happens

Alex-DeIt, 2024-05-20:
No, snipah,  I am not sure. Thought that dummy was a sort of generic interface.