fl4p/batmon-ha #233: Intallation error

candidotsa, 2024-06-01:
hi. 

i'm trying to intall the add-on but i get an error. 

The command '/bin/ash -o pipefail -c venv/bin/pip3 install -r requirements.txt' returned a non-zero code: 1

its my home assistant or is the add-on with a bug? 

thanks to all

END
```

Goaheadz, 2024-06-03:
I have the same issue.
Any solution?

candidotsa, 2024-06-03:
i think that the problem is a missing "b", but i don't know how to change and test.
'/bin/ash -o pipefail 
/bin/bash -o pipefail

Goaheadz, 2024-06-04:
[image]
[image]

no /bin/ash can be found in files

aleks2910, 2024-06-13:
I have the same issue. ((
Any solution?

candidotsa, 2024-06-22:
hi. its working :) thanks

closed 2024-06-22 by candidotsa