fl4p/batmon-ha #235: JK BMS stop reporting Voltage, Amperes, Soc, etc. Report most of the remain values without problems.

Delphinus44, 2024-06-19:
A couple of months ago 1 of my 3 BMS stop reporting V, A, W, charge but the report the remain values, this week a second BMS start with same behavior, at the moment only one of the BMS reports 100% of the values, there was no change in BMS, Batmon and Hassio.

Tobfaer, 2024-06-23:
I have the same issue since a couple of days. First it was intermittend, now its permanent. SOC, W, A, are no longer being displayed, all values are zero. Voltages however (min, max, average, min cell, max cell....) are completely fine and continue to be displayed. No changes were done to the system. In the JKBMS app all values are correctly reported.

**Logs:**
21:34:00 INFO [bt] BT Discovery: 
21:34:00 INFO [bt] BT Discovery: 
21:34:06 INFO [bt] BT C4:47:8C:EA:C3:F8                JK_B1A8S20P 
21:34:06 INFO [jikong] normal connect failed ([org.bluez.Error.Failed] Input/output error), connecting with scanner 
21:34:39 INFO [sampling] connected bms JKBt(C4:47:8C:EA:C3:F8,JKBMS)! 
21:34:40 INFO [sampling] Sending HA discovery for JKBMS (num_samples=1237) 
21:38:03 INFO [sampling] JKBMS: BmsSampl(0.0%,U=0.0V,I=0.00A,P=0W,Q=17826/0Ah,mos=0°C) 
21:38:03 INFO [sampling] JKBMS volt=[3321,3321,3322,3321,3320,3320,3320,3321] temp=[0.0, 0.0] 
21:39:41 INFO [sampling] Sending HA discovery for JKBMS (num_samples=1513) 
21:44:42 INFO [sampling] Sending HA discovery for JKBMS (num_samples=1788) 
21:48:05 INFO [sampling] JKBMS: BmsSampl(0.0%,U=0.0V,I=0.00A,P=0W,Q=17760/0Ah,mos=0°C) 
21:48:05 INFO [sampling] JKBMS volt=[3323,3325,3323,3326,3322,3323,3323,3323] temp=[0.0, 0.0] 
21:49:43 INFO [sampling] Sending HA discovery for JKBMS (num_samples=2063) 
21:53:06 INFO [sampling] JKBMS: BmsSampl(0.0%,U=0.0V,I=0.00A,P=0W,Q=17760/0Ah,mos=0°C) 
21:53:06 INFO [sampling] JKBMS volt=[3325,3325,3325,3323,3322,3323,3322,3322] temp=[0.0, 0.0]

haselnuss1971, 2024-07-10:
i have the same issue. One of 3 BMS has Problems...
07:34:42 INFO [sampling] Block3 subscribing for {'charge': False, 'discharge': False, 'balance': True} switch change
07:34:42 INFO [sampling] Block3: BmsSampl(0.0%,U=0.0V,I=0.00A,P=0W,Q=4163895/0Ah,mos=nan°C)
07:34:42 INFO [sampling] Block3 volt=[3324,3327,3326,3326,3326,3326,3327,3326,3328,3327,3327,3327,3326,3327,3327,3326] temp=[nan, nan]

haselnuss1971, 2024-07-11:
for completeness:
In the App (JK), all values are displayed correct.

alinfulger, 2024-07-12:
Easiest workaround is to connect with the app to JK BMS and just modify Capacity to different number and then back to real capacity. For me, this has restored the functionality of Voltage / Ampere / Power readings.

esther1112, 2024-07-30:
> Easiest workaround is to connect with the app to JK BMS and just modify Capacity to different number and then back to real capacity. For me, this has restored the functionality of Voltage / Ampere / Power readings.

This worked for us. Don't turn the BMS off or on, just do amp hour settings via app as instructed, all figure back to what they should be now. Thankyou

fl4p, 2025-02-02:
Hi, please update to 1.81 and try either jk_24s or jk_32s instead of jk as BMS type

closed 2025-02-02 by fl4p