fl4p/batmon-ha #237: Batmon intermittently working

jcottis, 2024-07-06:
I have had Batmon working intermittently working for a couple of days now however it stops working and I am not sure why. I've restarted several times, tried different settings for keep_alive and bt_power_cycle however it doesn't start. I look a couple hours later and then suddenly it's working again then stops after say 30 minutes, I am using a RPi 3 Model B+. Any assistance is appreciated

**Log File**
7:35:46 INFO [bt] BT C5:7D:5B:94:CA:96         Winch Batt Charger
07:35:46 INFO [bt] BT 49:15:BB:29:F9:68          49-15-BB-29-F9-68
07:35:46 INFO [bt] BT 24:4B:03:3D:A8:6C          24-4B-03-3D-A8-6C
07:35:46 INFO [main] Bleak version 0.13.1a1, BtBackend version bluez-v5.76
07:35:46 INFO [main] connecting mqtt addons@core-mosquitto
/app/main.py:196: DeprecationWarning: Callback API version 1 is deprecated, update to latest version
  mqtt_client = paho.Client(paho.CallbackAPIVersion.VERSION1)
07:35:46 INFO [main] Fetching 1 BMS + 0 virtual + 0 others serially, period=10.00s, keep_alive=True
07:35:46 INFO [sampling] connecting bms DalyBt(40:16:09:01:4F:05,House Bank)
07:35:46 INFO [main] mqtt watchdog loop started with timeout 300.0s
07:35:56 ERROR [bt] Device with address 40:16:09:01:4F:05 was not found., starting scanner
07:35:56 INFO [bt] BT Discovery:
07:36:01 INFO [bt] BT EE:44:42:8F:32:41        Smart Battery Start
07:36:01 INFO [bt] BT D0:03:DF:A8:5A:D3          D0-03-DF-A8-5A-D3
07:36:01 INFO [bt] BT E4:06:EB:6A:B2:9D        Smart Battery House
07:36:01 INFO [bt] BT CA:79:1A:04:E7:BA       Phoenix Smartcharger

**Configuration**
- address: 40:16:09:01:4F:05
  type: daly
  alias: House Bank

jcottis, 2024-07-06:
Just started working again, see attached screenshot and Batmon log file, any assistance appreciated.
[image]
[Batmon Log File whilst its working.txt](https://github.com/user-attachments/files/16117515/Batmon.Log.File.whilst.its.working.txt)

fl4p, 2026-09-03:
The scan lists other devices but not the Daly module, so the radio works and the BMS is simply not advertising at that moment. The Daly BT module accepts one connection and stops advertising while a phone with the Daly/Smart BMS app is connected or auto-reconnecting, and it also goes quiet when the BMS sleeps at low current. Close the app on all phones (or forget the device) and see if the dropouts stop. No activity since 2024, closing; reopen with a verbose_log from a current version if it persists with the app off.

closed 2026-09-03 by fl4p