fl4p/batmon-ha #238: JK inverter BMS: connection losses after 1 week + depreciation warning + other issues in the log (like "Unable to open mgmt_socket")

typxxi, 2024-07-07:
Pi 4 connected via Ethernet and BT to the JK BMS (inside an EEL Case)
Pi 4 sits on top of the case about 10 or 15 cm away from the BMS inside the metal box
HA OS latest June release "2024.6.4"
Mosquitto broker  6.4.1 
Usually the battery is connected to a laptop running windows and JK BMS Software plus VNC which I use to cross check if my inverter has reacted cause the growatt inverter is connected via cloud and that updates only every minute while I can instantly check the BMS "reaction". 
So this Laptop is connected via RS485 modbus and a USB cable and usually running JK BMS software 247.

**BatMon-HA Setup right now**
[image]

Here is the depreciation warning (but my mosquitto is the most recent one, no update open)

connecting mqtt addons@core-mosquitto
/app/main.py:196: **DeprecationWarning: Callback API version 1 is deprecated, update to latest version
  mqtt_client = paho.Client(paho.CallbackAPIVersion.VERSION1**) 

I have no clue what this means cause usually the mosquitto would offer me an update  but there is none available. 

```
Successfully installed bleak-0.20.2 dbus-fast-1.95.2
20:08:24 INFO [main] Power cycle bluetooth hardware
Unable to open mgmt_socket
Unable to open mgmt_socket
20:08:27 INFO [bt] BT Discovery:
20:08:32 INFO [bt] BT 01:C8:C5:EE:76:71          01-C8-C5-EE-76-71
20:08:32 INFO [bt] BT C8:47:80:10:08:A1           HBB_PB2A16S20-01
20:08:32 INFO [main] Bleak version 0.20.2, BtBackend version bluez-v5.76
20:08:32 INFO [main] connecting mqtt addons@core-mosquitto
/app/main.py:196: DeprecationWarning: Callback API version 1 is deprecated, update to latest version
  mqtt_client = paho.Client(paho.CallbackAPIVersion.VERSION1)
20:08:32 INFO [main] Fetching 1 BMS + 0 virtual + 0 others serially, period=1.00s, keep_alive=True
20:08:32 INFO [sampling] connecting bms JKBt(C8:47:80:10:08:A1,battery1)
20:08:32 INFO [main] mqtt watchdog loop started with timeout 300.0s
```

Looks like there is still a connection issue somehow even though it had been working  in the beginning quite well over days , I mean I have watched the figures and graphs over hours at the beginning and tweaked a bit to 2 sec sample and 3 or 4 sec mqqt but usually the temperatures are missing quite often while most or all others are available.

But the data are there cause I can follow those via a windows PC running the JK BMS (via vnc).

Here is the protocol from the moment I stopped and then restarted the add on

```
20:25:24 INFO [sampling] connecting bms JKBt(C8:47:80:10:08:A1,battery1)
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped
20:25:26 INFO [main] exit signal handler... (15, <frame at 0x7f9e6fc440, file '/usr/lib/python3.12/selectors.py', line 468, code select>), {}, shutdown was False
20:25:26 INFO [sampling] connected bms JKBt(C8:47:80:10:08:A1,battery1)!
20:25:27 INFO [main] fetch_loop <function main.<locals>.fn at 0x7f9e7004a0> ends
20:25:27 INFO [main] All fetch loops ended. shutdown is already 1
20:25:27 INFO [main] Disconnecting JKBt(C8:47:80:10:08:A1,battery1)
20:25:28 INFO [main] Background thread ends. shutdown=True
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
20:25:51 INFO [install_bleak] bleak 0.13.1a1 installed, want 0.20.2, running pip3 install bleak==0.20.2
20:25:51 INFO [install_bleak] influxdb installed=False need=False
Looking in indexes: https://pypi.org/simple, https://wheels.home-assistant.io/musllinux-index/
Collecting bleak==0.20.2
  Downloading https://wheels.home-assistant.io/musllinux-index/bleak-0.20.2-py3-none-any.whl (135 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 135.6/135.6 kB 468.2 kB/s eta 0:00:00
Collecting dbus-fast<2.0.0,>=1.83.0 (from bleak==0.20.2)
  Downloading dbus_fast-1.95.2.tar.gz (67 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 67.9/67.9 kB 3.7 MB/s eta 0:00:00
  Installing build dependencies: started
  Installing build dependencies: finished with status 'done'
  Getting requirements to build wheel: started
  Getting requirements to build wheel: finished with status 'done'
  Preparing metadata (pyproject.toml): started
  Preparing metadata (pyproject.toml): finished with status 'done'
Building wheels for collected packages: dbus-fast
  Building wheel for dbus-fast (pyproject.toml): started
  Building wheel for dbus-fast (pyproject.toml): finished with status 'done'
  Created wheel for dbus-fast: filename=dbus_fast-1.95.2-cp312-cp312-musllinux_1_2_aarch64.whl size=1054812 sha256=f87f5f86be1edf2ee9ffd6df6ccfd6affaa5fd6065bff1f7071806757603699b
  Stored in directory: /tmp/pip-ephem-wheel-cache-ziw4b5m1/wheels/85/d3/15/64c15063630c6ab394749c963dc8b03f30cdc629f08ae943b0
Successfully built dbus-fast
Installing collected packages: dbus-fast, bleak
  Attempting uninstall: bleak
    Found existing installation: bleak 0.13.1a1
    Uninstalling bleak-0.13.1a1:
      Successfully uninstalled bleak-0.13.1a1
Successfully installed bleak-0.20.2 dbus-fast-1.95.2
20:26:21 INFO [main] Power cycle bluetooth hardware
Unable to open mgmt_socket
Unable to open mgmt_socket
20:26:24 INFO [bt] BT Discovery:
20:26:29 INFO [bt] BT 11:C8:55:83:74:54          11-C8-55-83-74-54
20:26:29 INFO [bt] BT C8:47:80:10:08:A1           HBB_PB2A16S20-01
20:26:29 INFO [main] Bleak version 0.20.2, BtBackend version bluez-v5.76
20:26:29 INFO [main] connecting mqtt addons@core-mosquitto
/app/main.py:196: DeprecationWarning: Callback API version 1 is deprecated, update to latest version
  mqtt_client = paho.Client(paho.CallbackAPIVersion.VERSION1)
20:26:29 INFO [main] Fetching 1 BMS + 0 virtual + 0 others serially, period=1.00s, keep_alive=True
20:26:29 INFO [sampling] connecting bms JKBt(C8:47:80:10:08:A1,battery1)
20:26:29 INFO [main] mqtt watchdog loop started with timeout 300.0s
20:26:32 INFO [sampling] connected bms JKBt(C8:47:80:10:08:A1,battery1)!
20:26:32 INFO [sampling] battery1 subscribing for {'charge': True, 'discharge': True, 'balance': True} switch change
20:26:32 INFO [sampling] battery1: BmsSampl(100.0%,U=53.3V,I=0.00A,P=0W,Q=311/311Ah,mos=nan°C)
20:26:32 INFO [sampling] battery1 volt=[3329,3327,3328,3328,3328,3328,3328,3329,3328,3328,3328,3329,3329,3330,3328,3328] temp=[nan, nan, nan, nan]
20:26:32 INFO [sampling] Sending HA discovery for battery1 (num_samples=0)
```

The verbose log is much longer and detailed but since I have activated that roughly 15 minutes ago the data arrived more consistantly and I had no further losses. 

30 minutes later and still constant figures and much, much less "unavailable" messages (only the temperatures are missing, but only very rarely or maybe only once in 3o minutes). 

I am not sure what is going on but to let you know: the battery is over 30 meters away in a closed cellar room with 25°C and no one has been there. Also no interferences possible.

typxxi, 2024-07-07:
Just to let you know for whatever reason now the connection seems to be stable

[image]

I have not been in the room for 3 or 4  hours and no one else. I have only activated the verbose log.

typxxi, 2024-07-08:
Final Update: the verbose log has NO ISSUE and the losses came back this morning when I figured out that the door of the inverter room had been closed.
So the full metall door seems to have an impact if it is open or not which means the reflection of bt waves must have caused a better availability and connection between Raspi Pi and its onboard BT and the EEL V4 metal battery box with JK BMS inside.

Here is the graph for the last week.

[image]

The open door stops the reflection and breaks the connection this way.

fl4p, 2024-07-14:
Hi,
I guess you have an inverter running, what is its brand?

In linux you can usually disable Wifi with
```
rfkill block wifi
```

This might help you for HA OS:
https://community.home-assistant.io/t/disabling-built-in-wi-fi-bluetooth-from-raspberry-pi-cm4-on-home-assistant-yellow/644312/10

typxxi, 2024-07-31:
> This might help you for HA OS:
> https://community.home-assistant.io/t/disabling-built-in-wi-fi-bluetooth-from-raspberry-pi-cm4-on-home-assistant-yellow/644312/10

Yes - and no
cause I am unsure if this description allows to turn off WiFi only and let Bluetooth turned on.
I have seen that there were 2 lines , 1 for wifi and 1 for bt, but I am unsure if that will work out for WiFi only.

Any experiences what to do for turning WiFi only off ?
My Pi4 is connected via LAN cable , but of cause BATmon is relying on BT. 
Thanks

ilikenwf, 2025-07-02:
What I finally got working was disabling the watchddog checkbox on the configuration page. I also updated to bleak 1.0.1 but I'm not sure that really did anything.