fl4p/batmon-ha #240: Error discovering devices

kmoreau893, 2024-07-16:
I'm a Solar assistant newbie, and on top of it I'm running it on MacOS VMWare / Linux with a few USB Bluetooth adapters, but I think that part is working. I installed MQTT and Batmon, but getting this error right on startup:

19:27:42 INFO [bt] BT Discovery:
19:27:42 ERROR [main] Error discovering devices: [org.freedesktop.DBus.Error.UnknownObject] Method "SetDiscoveryFilter" with signature "a{sv}" on interface "org.bluez.Adapter1" doesn't exist

19:27:42 INFO [main] Verbose logging enabled
19:27:42 INFO [main] Bleak version 0.13.1a1, BtBackend version bluez-v5.76
19:27:42 WARNING [__init__] Unknown device type {'address': '69:8C:21:37:C1:A4', 'type': 'JBD', 'alias': 'Lishen_273_Boxed'}
19:27:42 INFO [main] Skip {'address': '69:8C:21:37:C1:A4', 'type': 'JBD', 'alias': 'Lishen_273_Boxed'}
19:27:42 INFO [main] connecting mqtt addons@core-mosquitto
/app/main.py:196: DeprecationWarning: Callback API version 1 is deprecated, update to latest version
  mqtt_client = paho.Client(paho.CallbackAPIVersion.VERSION1)
19:27:42 INFO [main] Fetching 0 BMS + 0 virtual + 0 others serially, period=1.00s, keep_alive=True
19:27:42 INFO [main] mqtt watchdog loop started with timeout 300.0s

Thanks for any advice! -Keith

closed 2024-07-17 by kmoreau893

Satsuger, 2024-11-14:
@kmoreau893 any success with this issue?
i'm having the same

kmoreau893, 2024-11-14:
@Satsuger :I can't remember how it fixed itself, perhaps with a different bluetooth adapter or restarting the whole batman service or bluetooth service, something like that within Home Assistant.