fl4p/batmon-ha #244: Offline usage

slundell, 2024-07-22:
When the addon starts, it checks if it needs to install a new bleak or influxdb using pip. This requires internet connection. If possible, this package should be cached, or preinstalled - or should maybe continue to run but with default package versions.

fl4p, 2024-07-22:
Yes, I am working on this. batmon uses an older bleak fork that has a proper pairing agent, and the environment files in the Docker will be reset every time the container starts, so it needs to re-download bleak every time. I am about to create a 2nd virtual env for the older version.

Influxdb should be added to the Dockerfile .

fl4p, 2024-07-22:
Also, batmon should probably just continue even if pip fails

fl4p, 2025-02-02:
hi, this has been fixed with 1.81 https://github.com/fl4p/batmon-ha/commit/af04ba33828bf113dfc022c92de77874deba10fe

closed 2025-02-02 by fl4p