fl4p/batmon-ha #246: Unable to find or connect to Daly BMSs

Robifz, 2024-07-28:
Thanks for the Add-on. I would really like to get it to work

This is my config
- address: "18:81:03:01:00:27"
  type: daly
  alias: lifepo4-1
- address: "18:81:03:05:01:96"
  type: daly
  alias: lifepo4-2

I am using a Bluetooth scanner to find the MAC adddress.
I have a Rpi 4 with HAOS
BMSs are less than 2m from the Rpi
With the DALY BNS app I can always connect from a distance of at least 5m
Is the Rpi BT so weak??
The discovery only finds one ESP32 that is 1m from the Rpi.
I also have many other ESP32s that the discovery does not find either.

BEGIN 
01:38:52 INFO [bt] BT Discovery:
01:38:58 INFO [bt] BT 40:22:D8:3E:9A:62                  esp32-n2k
01:38:58 INFO [daly] normal connect failed (Device with address 18:81:03:05:01:96 was not found.), connecting with scanner
01:39:07 ERROR [sampling] DalyBt(18:81:03:05:01:96,lifepo4-2) device not found, retry in 1 seconds
01:39:07 INFO [sampling] connecting bms DalyBt(18:81:03:01:00:27,lifepo4-1)
01:39:17 ERROR [bt] Device with address 18:81:03:01:00:27 was not found., starting scanner
01:39:17 INFO [bt] BT Discovery:
01:39:22 INFO [bt] BT 40:22:D8:3E:9A:62                  esp32-n2k
01:39:22 INFO [daly] normal connect failed (Device with address 18:81:03:01:00:27 was not found.), connecting with scanner
01:39:32 ERROR [sampling] DalyBt(18:81:03:01:00:27,lifepo4-1) device not found, retry in 1 seconds
01:39:36 INFO [sampling] connecting bms DalyBt(18:81:03:05:01:96,lifepo4-2)
01:39:46 ERROR [bt] Device with address 18:81:03:05:01:96 was not found., starting scanner

END
```

fl4p, 2025-02-02:
Try to re-connect the BMS dongle and make sure its not too far away from your home assistant host.

fat-fred, 2025-04-08:
for me its the same with the BT/WIFI Modul

If i power off/on the BMS, i can see it for ca. 1 sec. Batmon tries to connect, but fails. 
A Bluetooth Scanner on Android can also not more see it.

But with the "smart BMS" App i can connect to the BMS.

Thats the log in the moment i power on the BMS:

`18:53:58 INFO [daly] battery normal connect failed (Device with address 7C:3E:82:1C:DD:4E was not found.), connecting with scanner
18:54:08 ERROR [sampling] DalyBt(7C:3E:82:1C:DD:4E,battery) device not found, retry in 1 seconds
18:54:13 INFO [sampling] connecting bms DalyBt(7C:3E:82:1C:DD:4E,battery)
18:54:24 WARNING [daly] tried rx/tx/sx uuids 17/15/48: Characteristic 48 was not found!
18:54:24 WARNING [daly] tried rx/tx/sx uuids 0000fff1-0000-1000-8000-00805f9b34fb/0000fff2-0000-1000-8000-00805f9b34fb/02f00000-0000-0000-0000-00000000ff01: Characteristic 02f00000-0000-0000-0000-00000000ff01 was not found!
18:54:24 INFO [bt] [Service] 00001801-0000-1000-8000-00805f9b34fb (Handle: 1): Generic Attribute Profile
18:54:24 INFO [bt] 	[Characteristic] 00002a05-0000-1000-8000-00805f9b34fb (Handle: 2): Service Changed (indicate), Value: None
18:54:24 INFO [bt] 		[Descriptor] 00002902-0000-1000-8000-00805f9b34fb (Handle: 4): Client Characteristic Configuration) | Value: b'\x02\x00'
18:54:24 INFO [bt] [Service] 0000fff0-0000-1000-8000-00805f9b34fb (Handle: 16): Vendor specific
18:54:24 INFO [bt] 	[Characteristic] 0000fff2-0000-1000-8000-00805f9b34fb (Handle: 20): Vendor specific (read,write-without-response,write), Value: b'\x02\x00-F7C3E821CDD4E\x00\x02\t\x00'
18:54:24 WARNING [daly] checksum fail, expected 68, got 69. bytearray(b'7C3E821C')
18:54:24 WARNING [daly] msg too short: bytearray(b'D4E\x00\x02\t')
18:54:24 INFO [bt] 	[Characteristic] 0000fff1-0000-1000-8000-00805f9b34fb (Handle: 17): Vendor specific (read,notify), Value: b'\x02\x00-F7C3E821CDD4E\x00\x02\t'
18:54:24 INFO [bt] 		[Descriptor] 00002902-0000-1000-8000-00805f9b34fb (Handle: 19): Client Characteristic Configuration) | Value: b'\x01\x00'
18:54:25 INFO [bt] 	[Characteristic] 0000fffb-0000-1000-8000-00805f9b34fb (Handle: 24): Vendor specific (read,write-without-response,write,notify), Value: b'\x01\x00-F7C3E821CDD4E\x00\x02\t\x00'
18:54:25 INFO [bt] 	[Characteristic] 0000fff3-0000-1000-8000-00805f9b34fb (Handle: 26): Vendor specific (read,write,notify), Value: b'\x01\x00-F7C3E821CDD4E\x00\x02\t\x00\x01*'
18:54:25 INFO [bt] 		[Descriptor] 00002902-0000-1000-8000-00805f9b34fb (Handle: 28): Client Characteristic Configuration) | Value: b'\x00\x00'
18:54:25 INFO [bt] 	[Characteristic] 0000fffa-0000-1000-8000-00805f9b34fb (Handle: 22): Vendor specific (read,write-without-response,write), Value: b'\x01\x00'
18:54:25 INFO [bt] [Service] 0000ff00-0000-1000-8000-00805f9b34fb (Handle: 10): Vendor specific
18:54:25 INFO [bt] 	[Characteristic] 0000ff02-0000-1000-8000-00805f9b34fb (Handle: 14): Vendor specific (read,write-without-response,write), Value: b''
18:54:25 INFO [bt] 	[Characteristic] 0000ff01-0000-1000-8000-00805f9b34fb (Handle: 11): Vendor specific (read,notify), Value: b'\x01\x00-F7C3E821CDD4E\x00\x02\t\x00\x01*\x0b\x00\x12\x0c\x00\x01\xff\x0e'
18:54:25 INFO [bt] 		[Descriptor] 00002902-0000-1000-8000-00805f9b34fb (Handle: 13): Client Characteristic Configuration) | Value: b'\x00\x00'
18:54:25 ERROR [sampling] battery error (#3): Notify characteristic (rx) not found
Traceback (most recent call last):
  File "/app/bmslib/sampling.py", line 155, in __call__
    s = await self._sample_inner()
        ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/bmslib/sampling.py", line 229, in _sample_inner
    async with bms:
               ^^^
  File "/app/bmslib/bt.py", line 348, in __aenter__
    await self.connect()
  File "/app/bmslib/models/daly.py", line 147, in connect
    raise Exception("Notify characteristic (rx) not found")
Exception: Notify characteristic (rx) not found
18:54:25 INFO [sampling] battery bms debug data: {'r': None, 'buf': {}, 'rx': None, 'tx': None}
18:54:25 INFO [sampling] Bleak version 0.20.2
18:54:25 ERROR [main] 1 exceptions occurred fetching BMSs
18:54:25 ERROR [main] Error (num 1, max 200) reading BMS: Notify characteristic (rx) not found
18:54:25 ERROR [main] Stack: Traceback (most recent call last):
  File "/app/main.py", line 37, in fetch_loop
    if await fn():
       ^^^^^^^^^^
  File "/app/main.py", line 331, in fn
    raise exceptions[0]
  File "/app/main.py", line 326, in fn
    await t()
  File "/app/bmslib/sampling.py", line 155, in __call__
    s = await self._sample_inner()
        ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/bmslib/sampling.py", line 229, in _sample_inner
    async with bms:
               ^^^
  File "/app/bmslib/bt.py", line 348, in __aenter__
    await self.connect()
  File "/app/bmslib/models/daly.py", line 147, in connect
    raise Exception("Notify characteristic (rx) not found")
Exception: Notify characteristic (rx) not found
18:54:26 INFO [daly] battery normal connect failed (Client is already connected), connecting with scanner
18:54:29 ERROR [sampling] battery error (#4): <class 'EOFError'>
Traceback (most recent call last):
[... 201 lines cut ...]
    raise exceptions[0]
  File "/app/main.py", line 326, in fn
    await t()
  File "/app/bmslib/sampling.py", line 155, in __call__
    s = await self._sample_inner()
        ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/bmslib/sampling.py", line 229, in _sample_inner
    async with bms:
               ^^^
  File "/app/bmslib/bt.py", line 348, in __aenter__
[... cut ...]
  File "/app/venv/lib/python3.12/site-packages/bleak/backends/bluezdbus/scanner.py", line 191, in start
    self._stop = await manager.active_scan(
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/venv/lib/python3.12/site-packages/bleak/backends/bluezdbus/manager.py", line 368, in active_scan
    assert_reply(reply)
  File "/app/venv/lib/python3.12/site-packages/bleak/backends/bluezdbus/utils.py", line 22, in assert_reply
    raise BleakDBusError(reply.error_name, reply.body)
bleak.exc.BleakDBusError: [org.bluez.Error.InProgress] Operation already in progress`

fl4p, 2026-09-03:
Two things here. For the GATT dump in the second comment (`fff0` service with `fff1/fff2`, `Characteristic ... was not found`): that is the newer Daly Bluetooth module, use `type: daly2` (working since 2.03, #356). For the original report the adapter never saw the BMSes at all, along with several ESP32s nearby, so that is the RPi's Bluetooth reception; no reply since February 2025. Closing; open a new issue with a `verbose_log` if a current version still cannot see the device.

closed 2026-09-03 by fl4p