fl4p/batmon-ha #248: Unknown BMS BM100 values ​​going up and down in one volt increments

Ordiproxi, 2024-08-09:
Good morning,

The values ​​spend their time going up and down for example they go from 11.50v to 12.50 volt in increments of 1 volt once they reach 12.50 they go back down I used dummy because the other bms do not work the battery controller is a BM100 for car and 12.00 volt battery.
Bravo for your work and thank you
Describe the issue here.
If you experience bluetooth connection problems, follow these steps before opening an issue:
https://github.com/fl4p/batmon-ha#troubleshooting

* Include exact BMS model name
* Query HW and SW numbers from the BMS app
* Enable verbose_log

```
Paste log output between BEGIN and END:
BEGIN 
07:29:23 INFO [sampling] Sending HA discovery for Batterie 4 (num_samples=51302)
07:34:24 INFO [sampling] Sending HA discovery for Batterie 1 (num_samples=51602)
07:34:24 INFO [sampling] Sending HA discovery for Batterie 2 (num_samples=51602)
07:34:24 INFO [sampling] Sending HA discovery for Batterie 3 (num_samples=51602)
07:34:24 INFO [sampling] Sending HA discovery for Batterie 4 (num_samples=51602)
07:39:25 INFO [sampling] Sending HA discovery for Batterie 1 (num_samples=51902)
07:39:25 INFO [sampling] Sending HA discovery for Batterie 2 (num_samples=51902)
07:39:25 INFO [sampling] Sending HA discovery for Batterie 3 (num_samples=51902)
07:39:25 INFO [sampling] Sending HA discovery for Batterie 4 (num_samples=51902)
07:44:26 INFO [sampling] Sending HA discovery for Batterie 1 (num_samples=52202)
07:44:26 INFO [sampling] Sending HA discovery for Batterie 2 (num_samples=52202)
07:44:26 INFO [sampling] Sending HA discovery for Batterie 3 (num_samples=52202)
07:44:26 INFO [sampling] Sending HA discovery for Batterie 4 (num_samples=52202)
07:49:26 INFO [sampling] Sending HA discovery for Batterie 1 (num_samples=52502)
07:49:26 INFO [sampling] Sending HA discovery for Batterie 2 (num_samples=52502)
07:49:26 INFO [sampling] Sending HA discovery for Batterie 3 (num_samples=52502)
07:49:26 INFO [sampling] Sending HA discovery for Batterie 4 (num_samples=52502)
07:54:27 INFO [sampling] Sending HA discovery for Batterie 1 (num_samples=52802)
07:54:27 INFO [sampling] Sending HA discovery for Batterie 2 (num_samples=52802)
07:54:27 INFO [sampling] Sending HA discovery for Batterie 3 (num_samples=52802)
07:54:27 INFO [sampling] Sending HA discovery for Batterie 4 (num_samples=52802)
07:59:28 INFO [sampling] Sending HA discovery for Batterie 1 (num_samples=53102)
07:59:28 INFO [sampling] Sending HA discovery for Batterie 2 (num_samples=53102)
07:59:28 INFO [sampling] Sending HA discovery for Batterie 3 (num_samples=53102)
07:59:28 INFO [sampling] Sending HA discovery for Batterie 4 (num_samples=53102)
08:04:29 INFO [sampling] Sending HA discovery for Batterie 1 (num_samples=53402)
08:04:29 INFO [sampling] Sending HA discovery for Batterie 2 (num_samples=53402)
08:04:29 INFO [sampling] Sending HA discovery for Batterie 3 (num_samples=53402)
08:04:29 INFO [sampling] Sending HA discovery for Batterie 4 (num_samples=53402)
08:09:30 INFO [sampling] Sending HA discovery for Batterie 1 (num_samples=53702)
08:09:30 INFO [sampling] Sending HA discovery for Batterie 2 (num_samples=53702)
08:09:30 INFO [sampling] Sending HA discovery for Batterie 3 (num_samples=53702)
08:09:30 INFO [sampling] Sending HA discovery for Batterie 4 (num_samples=53702)
08:14:30 INFO [sampling] Sending HA discovery for Batterie 1 (num_samples=54002)
08:14:30 INFO [sampling] Sending HA discovery for Batterie 2 (num_samples=54002)
08:14:30 INFO [sampling] Sending HA discovery for Batterie 3 (num_samples=54002)
08:14:30 INFO [sampling] Sending HA discovery for Batterie 4 (num_samples=54002)
08:19:31 INFO [sampling] Sending HA discovery for Batterie 1 (num_samples=54302)
08:19:31 INFO [sampling] Sending HA discovery for Batterie 2 (num_samples=54302)
08:19:31 INFO [sampling] Sending HA discovery for Batterie 3 (num_samples=54302)
08:19:31 INFO [sampling] Sending HA discovery for Batterie 4 (num_samples=54302)
08:24:32 INFO [sampling] Sending HA discovery for Batterie 1 (num_samples=54602)
08:24:32 INFO [sampling] Sending HA discovery for Batterie 2 (num_samples=54602)
08:24:32 INFO [sampling] Sending HA discovery for Batterie 3 (num_samples=54602)
08:24:32 INFO [sampling] Sending HA discovery for Batterie 4 (num_samples=54602)
08:29:33 INFO [sampling] Sending HA discovery for Batterie 1 (num_samples=54902)
08:29:33 INFO [sampling] Sending HA discovery for Batterie 2 (num_samples=54902)
08:29:33 INFO [sampling] Sending HA discovery for Batterie 3 (num_samples=54902)
08:29:33 INFO [sampling] Sending HA discovery for Batterie 4 (num_samples=54902)
08:34:33 INFO [sampling] Sending HA discovery for Batterie 1 (num_samples=55202)
08:34:33 INFO [sampling] Sending HA discovery for Batterie 2 (num_samples=55202)
08:34:33 INFO [sampling] Sending HA discovery for Batterie 3 (num_samples=55202)
08:34:34 INFO [sampling] Sending HA discovery for Batterie 4 (num_samples=55202)
08:39:34 INFO [sampling] Sending HA discovery for Batterie 1 (num_samples=55502)
08:39:34 INFO [sampling] Sending HA discovery for Batterie 2 (num_samples=55502)
08:39:34 INFO [sampling] Sending HA discovery for Batterie 3 (num_samples=55502)
08:39:34 INFO [sampling] Sending HA discovery for Batterie 4 (num_samples=55502)
08:44:35 INFO [sampling] Sending HA discovery for Batterie 1 (num_samples=55802)
08:44:35 INFO [sampling] Sending HA discovery for Batterie 2 (num_samples=55802)
08:44:35 INFO [sampling] Sending HA discovery for Batterie 3 (num_samples=55802)
08:44:35 INFO [sampling] Sending HA discovery for Batterie 4 (num_samples=55802)
08:49:36 INFO [sampling] Sending HA discovery for Batterie 1 (num_samples=56102)
08:49:36 INFO [sampling] Sending HA discovery for Batterie 2 (num_samples=56102)
08:49:36 INFO [sampling] Sending HA discovery for Batterie 3 (num_samples=56102)
08:49:36 INFO [sampling] Sending HA discovery for Batterie 4 (num_samples=56102)
08:54:37 INFO [sampling] Sending HA discovery for Batterie 1 (num_samples=56402)
08:54:37 INFO [sampling] Sending HA discovery for Batterie 2 (num_samples=56402)
08:54:37 INFO [sampling] Sending HA discovery for Batterie 3 (num_samples=56402)
08:54:37 INFO [sampling] Sending HA discovery for Batterie 4 (num_samples=56402)
08:59:37 INFO [sampling] Sending HA discovery for Batterie 1 (num_samples=56702)
08:59:37 INFO [sampling] Sending HA discovery for Batterie 2 (num_samples=56702)
08:59:37 INFO [sampling] Sending HA discovery for Batterie 3 (num_samples=56702)
08:59:37 INFO [sampling] Sending HA discovery for Batterie 4 (num_samples=56702)
09:04:38 INFO [sampling] Sending HA discovery for Batterie 1 (num_samples=57002)
09:04:38 INFO [sampling] Sending HA discovery for Batterie 2 (num_samples=57002)
09:04:38 INFO [sampling] Sending HA discovery for Batterie 3 (num_samples=57002)
09:04:38 INFO [sampling] Sending HA discovery for Batterie 4 (num_samples=57002)
09:09:39 INFO [sampling] Sending HA discovery for Batterie 1 (num_samples=57302)
09:09:39 INFO [sampling] Sending HA discovery for Batterie 2 (num_samples=57302)
09:09:39 INFO [sampling] Sending HA discovery for Batterie 3 (num_samples=57302)
09:09:39 INFO [sampling] Sending HA discovery for Batterie 4 (num_samples=57302)
09:14:40 INFO [sampling] Sending HA discovery for Batterie 1 (num_samples=57602)
09:14:40 INFO [sampling] Sending HA discovery for Batterie 2 (num_samples=57602)
09:14:40 INFO [sampling] Sending HA discovery for Batterie 3 (num_samples=57602)
09:14:40 INFO [sampling] Sending HA discovery for Batterie 4 (num_samples=57602)
09:19:40 INFO [sampling] Sending HA discovery for Batterie 1 (num_samples=57902)
09:19:40 INFO [sampling] Sending HA discovery for Batterie 2 (num_samples=57902)
09:19:40 INFO [sampling] Sending HA discovery for Batterie 3 (num_samples=57902)
09:19:40 INFO [sampling] Sending HA discovery for Batterie 4 (num_samples=57902)
09:24:41 INFO [sampling] Sending HA discovery for Batterie 1 (num_samples=58202)
09:24:41 INFO [sampling] Sending HA discovery for Batterie 2 (num_samples=58202)
09:24:41 INFO [sampling] Sending HA discovery for Batterie 3 (num_samples=58202)
09:24:41 INFO [sampling] Sending HA discovery for Batterie 4 (num_samples=58202)
09:29:42 INFO [sampling] Sending HA discovery for Batterie 1 (num_samples=58502)
09:29:42 INFO [sampling] Sending HA discovery for Batterie 2 (num_samples=58502)
09:29:42 INFO [sampling] Sending HA discovery for Batterie 3 (num_samples=58502)
09:29:42 INFO [sampling] Sending HA discovery for Batterie 4 (num_samples=58502)
09:34:43 INFO [sampling] Sending HA discovery for Batterie 1 (num_samples=58802)
09:34:43 INFO [sampling] Sending HA discovery for Batterie 2 (num_samples=58802)
09:34:43 INFO [sampling] Sending HA discovery for Batterie 3 (num_samples=58802)
09:34:43 INFO [sampling] Sending HA discovery for Batterie 4 (num_samples=58802)
END
```

fl4p, 2024-08-11:
the  `dummy` bms is intented for integration testing only, there is no real connection to a physical BMS, it just produces random data.

closed 2024-08-11 by fl4p

Ordiproxi, 2024-08-12:
Thank you for your answer for "dummy" I understood it after it was a virtual bms example.
I try this one:
Vgate BA100 Car Battery Tester Monitor 12V Bluetooth 4.0 Battery Assistant Works with iOS and Android Phones with 2023

https://fr.aliexpress.com/item/1005005227272827.html?spm=a2g0o.order_list.order_list_main.21.28ae5e5bj4Q2OF&gatewayAdapt=glo2fra

fragolinux, 2024-09-21:
> Thank you for your answer for "dummy" I understood it after it was a virtual bms example. I try this one: Vgate BA100 Car Battery Tester Monitor 12V Bluetooth 4.0 Battery Assistant Works with iOS [...]

hi @fl4p i've this same device, vgate ba100: https://www.aliexpress.com/item/1005006485246319.html
tried each protocol with no luck... is it supported? Phone app works just fine, device too
TIA

Ordiproxi, 2024-09-21:
hi @fl4p i've this same device, vgate ba100: https://www.aliexpress.com/item/1005006485246319.html
tried each protocol with no luck... is it supported? Phone app works just fine, device too
TIA

Hello, no it does not work with batmoon
Good day

fragolinux, 2024-09-21:
any chance it will be added in the future? any help i can give extracting whatever data, if you guide me? thanks

fl4p, 2025-02-04:
> hi [@fl4p](https://github.com/fl4p) i've this same device, vgate ba100: https://www.aliexpress.com/item/1005006485246319.html tried each protocol with no luck... is it supported? Phone app works just fine, device too TIA
> 
> Hello, no it does not work with batmoon Good day

Hi, if you are still interested, please open a new issue