fl4p/batmon-ha #251: How do i run this via a container with Home Assistant running in a container

ashipaek0, 2024-08-17:
like the caption says.

what is the image url?

how do i run this as a docker container along with home assistant running in a container?

fl4p, 2026-02-20:
Hi,
Why do you want everything to run in its own container?
I think its easier to not use containers, and just install python and the requirements, maybe in a virtual environment. It is not necessary to run the MQTT or Batmon in a container if HA does so. I don't see any advantage of containering here.

fl4p, 2026-09-03:
This has an official answer now: prebuilt multi-arch images at `ghcr.io/fl4p/batmon-ha` (`latest` or a pinned tag like `2.19`), with compose examples in [doc/Docker.md](https://github.com/fl4p/batmon-ha/blob/master/doc/Docker.md). No Dockerfile edits needed; MQTT credentials go via `MQTT_HOST`/`MQTT_USER`/`MQTT_PASSWORD` env vars or `options.json`. Mount the data directory, not just the file, so meter totals survive recreation. Closing.

closed 2026-09-03 by fl4p