fl4p/batmon-ha #264: 100balance BMS

as4tik, 2024-11-22:
### Problem statement
There is a one more type similar to daly BMS, but sold as another brand. 
I give a try to each type, which are available for the moment in the Add-on:
type can be jk, jbd, ant, daly, supervolt, sok, victron or dummy

and found, that non works, but "dummy".

Type "dummy" selected for the "100balance BMS" (a child of daily BMS) helps to receive the below result:
[image]

What is strange here, is that my battery is 8-cells pack (280Ah), not 4-cells (100Ah).
Also

### Discovery
Looks like if to copy-past dummy settings to 100balance and expand number of cells possible, may be will be a semi-ready settings for 100balance BMS.

I will try to do so by branch, but I am not proficient in coding and this will be my first attempt.

* Include exact BMS model name: "100balance" BMS 8-18s 200A with WiFi and Bluetooth
* Query HW and SW numbers from the BMS app: not sure, where I could take it
* Enable verbose_log: not sure, where I could take it

```
Paste log output between BEGIN and END:
BEGIN 
14:59:58 INFO [bt] BT Discovery:
15:00:03 INFO [bt] BT 24:EC:4A:9D:E1:06                  KMF530102
15:00:03 INFO [bt] BT 18:33:51:51:2D:7E          18-33-51-51-2D-7E
15:00:03 INFO [bt] BT 50:18:09:01:08:BB           JHB-5018090108BB
15:00:03 INFO [main] Bleak version 0.13.1a1, BtBackend version bluez-v5.76
15:00:03 INFO [main] connecting mqtt addons@core-mosquitto
/app/main.py:196: DeprecationWarning: Callback API version 1 is deprecated, update to latest version
  mqtt_client = paho.Client(paho.CallbackAPIVersion.VERSION1)
15:00:03 INFO [main] Fetching 2 BMS + 0 virtual + 0 others serially, period=10.00s, keep_alive=False
15:00:03 INFO [sampling] connecting bms DummyBt(50:18:09:01:08:BB,battery1)
15:00:03 INFO [sampling] connected bms DummyBt(50:18:09:01:08:BB,battery1)!
15:00:03 INFO [sampling] battery1 subscribing for {'charge': True, 'discharge': True} switch change
15:00:03 INFO [sampling] battery1: BmsSampl(74.8%,U=12.0V,I=0.06A,P=1W,Q=75/100Ah,mos=nan°C)
15:00:03 INFO [sampling] battery1 volt=[3000,3010,3020,3030] temp=[nan]
15:00:03 INFO [sampling] Sending HA discovery for battery1 (num_samples=0)
15:00:04 INFO [sampling] connecting bms DummyBt(24:EC:4A:9D:E1:06,shunt)
15:00:04 INFO [sampling] connected bms DummyBt(24:EC:4A:9D:E1:06,shunt)!
15:00:04 INFO [sampling] shunt subscribing for {'charge': True, 'discharge': True} switch change
END
```

argulp, 2024-12-07:
Hello, 
I'm not able to use batmon to monitor this BMS (in my case 100balance 80A  8-18s)

I have attached the logs, it seems the "protocol" have changed a little bit, but I can still using the daly android app to monitor this bms, thank you for your wonderfull jobs !

[2af0a32d_batmon_2024-12-07T10-27-01.783Z.log](https://github.com/user-attachments/files/18047977/2af0a32d_batmon_2024-12-07T10-27-01.783Z.log)

ftrueck, 2025-01-22:
Same issue with a JHB-24RTH BMS.

Software version: 60_240816_01T2
Hardware version: JHB-24RTH-V1.2

21:59:31 INFO [main] Bleak version 0.20.2, BtBackend version bluez-v5.79
21:59:31 INFO [main] connecting mqtt addons@core-mosquitto
21:59:31 INFO [main] Fetching 1 BMS + 0 virtual + 0 others serially, period=1.00s, keep_alive=True
21:59:31 INFO [sampling] connecting bms DalyBt(50:18:09:01:02:3E,battery1)
21:59:31 INFO [main] mqtt watchdog loop started with timeout 300.0s
21:59:36 WARNING [daly] tried rx/tx/sx uuids 17/15/48: Characteristic 17 not found!
21:59:36 INFO [sampling] connected bms DalyBt(50:18:09:01:02:3E,battery1)!
21:59:48 ERROR [sampling] battery1 error (#1): timeout awaiting result 93, got 0/1 responses

The 100balance belongs to daly. They bought off the brand and rebrand it to daly.

Rcscomp67, 2025-02-03:
I have the same problem in my case with a 100Balance 8-24s 200A. With type: daly and type: daly2 no connection is established. With type: dummy the BMS can be connected but with significant limitations. For example Cell Index max = 4 even though it is operated for 15s.