fl4p/batmon-ha #265: Wrong meter value in lasted update

leolionart, 2024-12-10:
JkBMS 2
[image]
This happen with all sensor in secondary battery. I have 2 jk bms and is fine with battery1 but battery2 is totally wrong

Confushion, 2024-12-13:
I had the same issue coming up out of the blue in version 1.78
Seems the 4th byte of the `cycle_capacity` field ( buf[189] )  is set to 0x2 in my case.
Looks like this byte value is slowly increasing...

This change fixed the issue for me:
 `is_new_11fw = buf[189] in {0x0, 0x1, 0x2} and buf[189 + 32] > 0  # 32 cell version`

leolionart, 2024-12-31:
The problem in the latest version is displaying the wrong battery percentage. It went up to 160000. I had to roll back to version 1.79

fl4p, 2025-02-02:
Hi, since version 1.81 you can manually set the jk version. either use type: jk_24s for an older model or jk_32s for a newer one. If you are unsure try either and see what works for you