fl4p/batmon-ha #266: Batmon only work with a maximum of 5 devices

Delphinus44, 2024-12-10:
Describe the issue here.

Have 6 batteries with JKBMS and a Victron Shunt, if I configure more than 5 devices have errors in connection and devices keep dropping, if I have only 5, Batmon works without problems, the log refer to the battery MB31-01 but if I remove one of the others this one connects and everything works.

* Include exact BMS model name
JK_PB2A16S20P
* Query HW and SW numbers from the BMS app
HW V15A
SW V15.24
* Enable verbose_log

```
Paste log output between BEGIN and END:
BEGIN 
13:09:23 INFO [sampling] connecting bms JKBt(C8:47:80:14:01:46,MB31-01)
13:09:25 INFO [jikong] normal connect failed ([org.bluez.Error.InProgress] Operation already in progress), connecting with scanner
13:09:35 ERROR [sampling] MB31-01 error (#1): [org.bluez.Error.InProgress] Operation already in progress
Traceback (most recent call last):
  File "/app/bmslib/models/jikong.py", line 127, in connect
    await super().connect(timeout=6)
  File "/app/bmslib/bt.py", line 237, in connect
    await self._connect_client(timeout=timeout)
  File "/app/bmslib/bt.py", line 186, in _connect_client
    await asyncio.wait_for(self.client.connect(timeout=timeout), timeout=timeout + 1)
  File "/usr/lib/python3.12/asyncio/tasks.py", line 520, in wait_for
    return await fut
           ^^^^^^^^^
  File "/app/venv/lib/python3.12/site-packages/bleak/__init__.py", line 531, in connect
    return await self._backend.connect(**kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/venv/lib/python3.12/site-packages/bleak/backends/bluezdbus/client.py", line 124, in connect
    device = await BleakScanner.find_device_by_address(
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/venv/lib/python3.12/site-packages/bleak/__init__.py", line 310, in find_device_by_address
    return await cls.find_device_by_filter(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/venv/lib/python3.12/site-packages/bleak/__init__.py", line 369, in find_device_by_filter
    async with cls(detection_callback=apply_filter, **kwargs):
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/venv/lib/python3.12/site-packages/bleak/__init__.py", line 152, in __aenter__
    await self._backend.start()
  File "/app/venv/lib/python3.12/site-packages/bleak/backends/bluezdbus/scanner.py", line 191, in start
    self._stop = await manager.active_scan(
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/venv/lib/python3.12/site-packages/bleak/backends/bluezdbus/manager.py", line 368, in active_scan
    assert_reply(reply)
  File "/app/venv/lib/python3.12/site-packages/bleak/backends/bluezdbus/utils.py", line 22, in assert_reply
    raise BleakDBusError(reply.error_name, reply.body)
bleak.exc.BleakDBusError: [org.bluez.Error.InProgress] Operation already in progress
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
  File "/app/bmslib/sampling.py", line 155, in __call__
    s = await self._sample_inner()
        ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/bmslib/sampling.py", line 227, in _sample_inner
    async with bms:
               ^^^
  File "/app/bmslib/bt.py", line 345, in __aenter__
    await self.connect()
  File "/app/bmslib/models/jikong.py", line 130, in connect
    await self._connect_with_scanner(timeout=timeout)
  File "/app/bmslib/bt.py", line 273, in _connect_with_scanner
    await self._connect_client(timeout=timeout / 2)
  File "/app/bmslib/bt.py", line 186, in _connect_client
    await asyncio.wait_for(self.client.connect(timeout=timeout), timeout=timeout + 1)
  File "/usr/lib/python3.12/asyncio/tasks.py", line 520, in wait_for
    return await fut
           ^^^^^^^^^
  File "/app/venv/lib/python3.12/site-packages/bleak/__init__.py", line 531, in connect
    return await self._backend.connect(**kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/venv/lib/python3.12/site-packages/bleak/backends/bluezdbus/client.py", line 124, in connect
    device = await BleakScanner.find_device_by_address(
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/venv/lib/python3.12/site-packages/bleak/__init__.py", line 310, in find_device_by_address
    return await cls.find_device_by_filter(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/venv/lib/python3.12/site-packages/bleak/__init__.py", line 369, in find_device_by_filter
    async with cls(detection_callback=apply_filter, **kwargs):
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/venv/lib/python3.12/site-packages/bleak/__init__.py", line 152, in __aenter__
    await self._backend.start()
  File "/app/venv/lib/python3.12/site-packages/bleak/backends/bluezdbus/scanner.py", line 191, in start
    self._stop = await manager.active_scan(
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/venv/lib/python3.12/site-packages/bleak/backends/bluezdbus/manager.py", line 368, in active_scan
    assert_reply(reply)
  File "/app/venv/lib/python3.12/site-packages/bleak/backends/bluezdbus/utils.py", line 22, in assert_reply
    raise BleakDBusError(reply.error_name, reply.body)
bleak.exc.BleakDBusError: [org.bluez.Error.InProgress] Operation already in progress
13:09:35 INFO [sampling] MB31-01 bms debug data: {'resp': {}, 'char_w': None, 'char_r': None}
13:09:35 INFO [sampling] Bleak version 0.20.2
END
```

fl4p, 2024-12-28:
You might want to try adding another BLE adapter (USB dongle) and using the `adapter` paremeter to tell batmon which adapter to use.

fl4p, 2025-02-02:
Also i've updated bleak to 0.22, maybe that helps

fl4p, 2025-02-02:
I'll close this issue, re-open if needed (or create a new issue)

closed 2025-02-02 by fl4p