fl4p/batmon-ha #268: V1.80 Works only half jkbms

jariojose, 2024-12-31:
SOC, Power, current, voltagem, etc.. only show 0's values. I have 4 BMS with the same software version. Half of them show zero values. While the other half display correctly. This happened after updating to version 1.80.

JK_B2A8S20P
by JK
Firmware: 11.288
Hardware: 11.XW

Here is a wrong value :

```

JK_B2A8S20PCE: BmsSampl(0.0%,U=0.0V,I=0.00A,P=0W,Q=22610/0Ah,mos=0Â°C)
20:39:06 INFO [sampling] JK_B2A8S20PCE volt=[3316,3311,3315,3313,3308,3315,3314,3311] temp=[0.0, 0.0]
```

skosko1977, 2025-01-18:
I have the same problem in version 1.8, on version 1.79 everything works ok.

fl4p, 2025-02-02:
Hi, since version 1.81 you can manually set the jk version. either use type: `jk_24s` for an older model or `jk_32s` for a newer one. If you are unsure try either and see what works for you

fl4p, 2025-02-02:
dupe https://github.com/fl4p/batmon-ha/issues/265

closed 2025-02-02 by fl4p