fl4p/batmon-ha #269: batmon  fetching data  problem

xeongreg, 2025-01-02:
Describe the issue here.
If you experience bluetooth connection problems, follow these steps before opening an issue:
https://github.com/fl4p/batmon-ha#troubleshooting

* Include exact BMS model name
* Query HW and SW numbers from the BMS app
* Enable verbose_log

```
Paste log output between BEGIN and END:
BEGIN 
22:12:41 INFO [main] Bleak version 0.13.1a1, BtBackend version bluez-v5.79
22:12:41 INFO [main] connecting mqtt homeassistant@core-mosquitto
22:12:41 INFO [main] Fetching 1 BMS + 0 virtual + 0 others serially, period=2.00s, keep_alive=True
22:12:41 INFO [sampling] connecting bms JKBt(C8:47:80:1D:E8:14,battery1)
22:12:45 INFO [sampling] connected bms JKBt(C8:47:80:1D:E8:14,battery1)!
22:12:45 INFO [sampling] battery1 subscribing for {'charge': True, 'discharge': True, 'balance': True} switch change
22:12:45 WARNING [mqtt_util] mqtt publish battery1/soc/total_voltage failed: 4 (<MQTTErrorCode.MQTT_ERR_NO_CONN: 4>, 1)
22:12:45 WARNING [mqtt_util] mqtt publish battery1/soc/current failed: 4 (<MQTTErrorCode.MQTT_ERR_NO_CONN: 4>, 2)
22:12:45 WARNING [mqtt_util] mqtt publish battery1/soc/balance_current failed: 4 (<MQTTErrorCode.MQTT_ERR_NO_CONN: 4>, 3)
22:12:45 WARNING [mqtt_util] mqtt publish battery1/soc/soc_percent failed: 4 (<MQTTErrorCode.MQTT_ERR_NO_CONN: 4>, 4)
22:12:45 WARNING [mqtt_util] mqtt publish battery1/soc/power failed: 4 (<MQTTErrorCode.MQTT_ERR_NO_CONN: 4>, 5)
22:12:45 WARNING [mqtt_util] mqtt publish battery1/soc/capacity failed: 4 (<MQTTErrorCode.MQTT_ERR_NO_CONN: 4>, 6)
22:12:45 WARNING [mqtt_util] mqtt publish battery1/soc/cycle_capacity failed: 4 (<MQTTErrorCode.MQTT_ERR_NO_CONN: 4>, 7)
22:12:45 WARNING [mqtt_util] mqtt publish battery1/soc/num_cycles failed: 4 (<MQTTErrorCode.MQTT_ERR_NO_CONN: 4>, 8)
22:12:45 WARNING [mqtt_util] mqtt publish battery1/mosfet_status/capacity_ah failed: 4 (<MQTTErrorCode.MQTT_ERR_NO_CONN: 4>, 9)
22:12:45 WARNING [mqtt_util] mqtt publish battery1/bms/uptime failed: 4 (<MQTTErrorCode.MQTT_ERR_NO_CONN: 4>, 10)
22:12:45 WARNING [mqtt_util] mqtt publish battery1/meter/sample_count failed: 4 (<MQTTErrorCode.MQTT_ERR_NO_CONN: 4>, 11)
22:12:45 WARNING [mqtt_util] mqtt publish battery1/switch/charge failed: 4 (<MQTTErrorCode.MQTT_ERR_NO_CONN: 4>, 12)
22:12:45 WARNING [mqtt_util] mqtt publish battery1/switch/discharge failed: 4 (<MQTTErrorCode.MQTT_ERR_NO_CONN: 4>, 13)
22:12:45 WARNING [mqtt_util] mqtt publish battery1/switch/balance failed: 4 (<MQTTErrorCode.MQTT_ERR_NO_CONN: 4>, 14)
22:12:45 INFO [sampling] battery1: BmsSampl(36.5%,U=26.0V,I=-13.20A,P=-343W,Q=84/230Ah,mos=nan°C)
22:12:45 WARNING [mqtt_util] mqtt publish battery1/cell_voltages/1 failed: 4 (<MQTTErrorCode.MQTT_ERR_NO_CONN: 4>, 15)
22:12:45 WARNING [mqtt_util] mqtt publish battery1/cell_voltages/2 failed: 4 (<MQTTErrorCode.MQTT_ERR_NO_CONN: 4>, 16)
22:12:45 WARNING [mqtt_util] mqtt publish battery1/cell_voltages/3 failed: 4 (<MQTTErrorCode.MQTT_ERR_NO_CONN: 4>, 17)
22:12:45 WARNING [mqtt_util] mqtt publish battery1/cell_voltages/4 failed: 4 (<MQTTErrorCode.MQTT_ERR_NO_CONN: 4>, 18)
22:12:45 WARNING [mqtt_util] mqtt publish battery1/cell_voltages/5 failed: 4 (<MQTTErrorCode.MQTT_ERR_NO_CONN: 4>, 19)
22:12:45 WARNING [mqtt_util] mqtt publish battery1/cell_voltages/6 failed: 4 (<MQTTErrorCode.MQTT_ERR_NO_CONN: 4>, 20)
22:12:45 WARNING [mqtt_util] mqtt publish battery1/cell_voltages/7 failed: 4 (<MQTTErrorCode.MQTT_ERR_NO_CONN: 4>, 21)
22:12:45 WARNING [mqtt_util] mqtt publish battery1/cell_voltages/8 failed: 4 (<MQTTErrorCode.MQTT_ERR_NO_CONN: 4>, 22)
22:12:45 WARNING [mqtt_util] mqtt publish battery1/cell_voltages/min failed: 4 (<MQTTErrorCode.MQTT_ERR_NO_CONN: 4>, 23)
22:12:45 WARNING [mqtt_util] mqtt publish battery1/cell_voltages/min_index failed: 4 (<MQTTErrorCode.MQTT_ERR_NO_CONN: 4>, 24)
22:12:45 WARNING [mqtt_util] mqtt publish battery1/cell_voltages/max failed: 4 (<MQTTErrorCode.MQTT_ERR_NO_CONN: 4>, 25)
22:12:45 WARNING [mqtt_util] mqtt publish battery1/cell_voltages/max_index failed: 4 (<MQTTErrorCode.MQTT_ERR_NO_CONN: 4>, 26)
22:12:45 WARNING [mqtt_util] mqtt publish battery1/cell_voltages/delta failed: 4 (<MQTTErrorCode.MQTT_ERR_NO_CONN: 4>, 27)
22:12:45 WARNING [mqtt_util] mqtt publish battery1/cell_voltages/average failed: 4 (<MQTTErrorCode.MQTT_ERR_NO_CONN: 4>, 28)
22:12:45 WARNING [mqtt_util] mqtt publish battery1/cell_voltages/median failed: 4 (<MQTTErrorCode.MQTT_ERR_NO_CONN: 4>, 29)
22:12:45 INFO [sampling] battery1 volt=[3253,3250,3250,3248,3251,3253,3254,3254] temp=[nan, nan, nan, nan]
22:12:45 WARNING [mqtt_util] mqtt publish battery1/meter/total_charge failed: 4 (<MQTTErrorCode.MQTT_ERR_NO_CONN: 4>, 30)
22:12:45 WARNING [mqtt_util] mqtt publish battery1/meter/total_energy failed: 4 (<MQTTErrorCode.MQTT_ERR_NO_CONN: 4>, 31)
22:12:45 WARNING [mqtt_util] mqtt publish battery1/meter/total_energy_discharge failed: 4 (<MQTTErrorCode.MQTT_ERR_NO_CONN: 4>, 32)
22:12:45 WARNING [mqtt_util] mqtt publish battery1/meter/total_energy_charge failed: 4 (<MQTTErrorCode.MQTT_ERR_NO_CONN: 4>, 33)
22:12:45 WARNING [mqtt_util] mqtt publish battery1/meter/total_cycles failed: 4 (<MQTTErrorCode.MQTT_ERR_NO_CONN: 4>, 34)
22:12:45 WARNING [mqtt_util] mqtt publish battery1/meter/total_abs_diff_charge failed: 4 (<MQTTErrorCode.MQTT_ERR_NO_CONN: 4>, 35)
22:12:45 INFO [sampling] Sending HA discovery for battery1 (num_samples=0)
22:12:45 WARNING [mqtt_util] mqtt publish homeassistant/sensor/battery1/_soc_total_voltage/config failed: 4 (<MQTTErrorCode.MQTT_ERR_NO_CONN: 4>, 36)

END
```
22:12:45 INFO [sampling] battery1: BmsSampl(36.5%,U=26.0V,I=-13.20A,P=-343W,Q=84/230Ah,mos=nan°C)
22:12:45 INFO [sampling] battery1 volt=[3253,3250,3250,3248,3251,3253,3254,3254] temp=[nan, nan, nan, nan]

as you can see it gets some data from bms but not correct

xeongreg, 2025-01-02:
23:05:25 DEBUG [jikong] bms msg(22) (buf300): 0 0 0 0 0 0 0 0 1 fe ff 7f dc f 1 0 80 3 0 0 0 7e
23:05:25 DEBUG [jikong] got response 2 (len300)
23:05:25 DEBUG [jikong] New 11.x firmware, offset=32
23:05:25 WARNING [mqtt_util] mqtt publish battery1/soc/total_voltage failed: 4 (<MQTTErrorCode.MQTT_ERR_NO_CONN: 4>, 993)
23:05:25 WARNING [mqtt_util] mqtt publish battery1/soc/current failed: 4 (<MQTTErrorCode.MQTT_ERR_NO_CONN: 4>, 994)
23:05:25 WARNING [mqtt_util] mqtt publish battery1/soc/balance_current failed: 4 (<MQTTErrorCode.MQTT_ERR_NO_CONN: 4>, 995)
23:05:25 WARNING [mqtt_util] mqtt publish battery1/soc/soc_percent failed: 4 (<MQTTErrorCode.MQTT_ERR_NO_CONN: 4>, 996)
23:05:25 WARNING [mqtt_util] mqtt publish battery1/soc/power failed: 4 (<MQTTErrorCode.MQTT_ERR_NO_CONN: 4>, 997)
23:05:25 WARNING [mqtt_util] mqtt publish battery1/soc/capacity failed: 4 (<MQTTErrorCode.MQTT_ERR_NO_CONN: 4>, 998)
23:05:25 WARNING [mqtt_util] mqtt publish battery1/soc/cycle_capacity failed: 4 (<MQTTErrorCode.MQTT_ERR_NO_CONN: 4>, 999)
23:05:25 WARNING [mqtt_util] mqtt publish battery1/soc/num_cycles failed: 4 (<MQTTErrorCode.MQTT_ERR_NO_CONN: 4>, 1000)
23:05:25 WARNING [mqtt_util] mqtt publish battery1/mosfet_status/capacity_ah failed: 4 (<MQTTErrorCode.MQTT_ERR_NO_CONN: 4>, 1001)
23:05:25 WARNING [mqtt_util] mqtt publish battery1/mosfet_status/temperature failed: 4 (<MQTTErrorCode.MQTT_ERR_NO_CONN: 4>, 1002)
23:05:25 WARNING [mqtt_util] mqtt publish battery1/bms/uptime failed: 4 (<MQTTErrorCode.MQTT_ERR_NO_CONN: 4>, 1003)
23:05:25 WARNING [mqtt_util] mqtt publish battery1/meter/sample_count failed: 4 (<MQTTErrorCode.MQTT_ERR_NO_CONN: 4>, 1004)
23:05:25 WARNING [mqtt_util] mqtt publish battery1/switch/charge failed: 4 (<MQTTErrorCode.MQTT_ERR_NO_CONN: 4>, 1005)
23:05:25 WARNING [mqtt_util] mqtt publish battery1/switch/discharge failed: 4 (<MQTTErrorCode.MQTT_ERR_NO_CONN: 4>, 1006)
23:05:25 WARNING [mqtt_util] mqtt publish battery1/switch/balance failed: 4 (<MQTTErrorCode.MQTT_ERR_NO_CONN: 4>, 1007)
**23:05:25 INFO [sampling] battery1: BmsSampl(32.5%,U=26.8V,I=59.38A,P=1590W,Q=75/230Ah,mos=27°C)**
23:05:25 WARNING [mqtt_util] mqtt publish battery1/cell_voltages/1 failed: 4 (<MQTTErrorCode.MQTT_ERR_NO_CONN: 4>, 1008)
23:05:25 WARNING [mqtt_util] mqtt publish battery1/cell_voltages/2 failed: 4 (<MQTTErrorCode.MQTT_ERR_NO_CONN: 4>, 1009)
23:05:25 WARNING [mqtt_util] mqtt publish battery1/cell_voltages/3 failed: 4 (<MQTTErrorCode.MQTT_ERR_NO_CONN: 4>, 1010)
23:05:25 WARNING [mqtt_util] mqtt publish battery1/cell_voltages/4 failed: 4 (<MQTTErrorCode.MQTT_ERR_NO_CONN: 4>, 1011)
23:05:25 WARNING [mqtt_util] mqtt publish battery1/cell_voltages/5 failed: 4 (<MQTTErrorCode.MQTT_ERR_NO_CONN: 4>, 1012)
23:05:25 WARNING [mqtt_util] mqtt publish battery1/cell_voltages/6 failed: 4 (<MQTTErrorCode.MQTT_ERR_NO_CONN: 4>, 1013)
23:05:25 WARNING [mqtt_util] mqtt publish battery1/cell_voltages/7 failed: 4 (<MQTTErrorCode.MQTT_ERR_NO_CONN: 4>, 1014)
23:05:25 WARNING [mqtt_util] mqtt publish battery1/cell_voltages/8 failed: 4 (<MQTTErrorCode.MQTT_ERR_NO_CONN: 4>, 1015)
23:05:25 WARNING [mqtt_util] mqtt publish battery1/cell_voltages/min failed: 4 (<MQTTErrorCode.MQTT_ERR_NO_CONN: 4>, 1016)
23:05:25 WARNING [mqtt_util] mqtt publish battery1/cell_voltages/min_index failed: 4 (<MQTTErrorCode.MQTT_ERR_NO_CONN: 4>, 1017)
23:05:25 WARNING [mqtt_util] mqtt publish battery1/cell_voltages/max failed: 4 (<MQTTErrorCode.MQTT_ERR_NO_CONN: 4>, 1018)
23:05:25 WARNING [mqtt_util] mqtt publish battery1/cell_voltages/max_index failed: 4 (<MQTTErrorCode.MQTT_ERR_NO_CONN: 4>, 1019)
23:05:25 WARNING [mqtt_util] mqtt publish battery1/cell_voltages/delta failed: 4 (<MQTTErrorCode.MQTT_ERR_NO_CONN: 4>, 1020)
23:05:25 WARNING [mqtt_util] mqtt publish battery1/cell_voltages/average failed: 4 (<MQTTErrorCode.MQTT_ERR_NO_CONN: 4>, 1021)
23:05:25 WARNING [mqtt_util] mqtt publish battery1/cell_voltages/median failed: 4 (<MQTTErrorCode.MQTT_ERR_NO_CONN: 4>, 1022)
**23:05:25 INFO [sampling] battery1 volt=[3341,3345,3346,3365,3353,3349,3345,3341] temp=[23.5, 23.2, 0.0, 0.0]**
23:05:25 INFO [sampling] JKBt(C8:47:80:1D:E8:14,battery1) times: connect=0.00s fetch=0.05s
23:05:25 ERROR [main] MQTT never published a message after 300s, exit

fl4p, 2026-09-03:
The BMS data was fine; the broker refused batmon's MQTT login, and batmon never said so, only "publish failed: 4" until the watchdog gave up. Since master it logs `MQTT broker ... refused the connection: Not authorized` at startup. With the Mosquitto add-on the credentials must be a Home Assistant user, not the add-on's own. Closing; open a new issue if a current version still cannot publish.

closed 2026-09-03 by fl4p