fl4p/batmon-ha #270: Installation error.

poligor84, 2025-01-10:
During the installation, I get the following error: The command '/bin/ash -o pipefail -c venv/bin/pip3 install -r requirements.txt' returned a non-zero code: 1
What do you advise me to do? Thank you.

Home Assistant
Core
2025.1.2
Supervisor
2024.12.3
Operating System
14.0
Інтерфейс
20250109.0

Логгер: homeassistant.components.hassio
Джерело: components/hassio/websocket_api.py:135
інтеграції: Home Assistant Supervisor (документація, проблеми)
Вперше відбулося: 07:34:25 (1 випадки)
Востаннє записано: 07:34:25

Failed to to call /addons/2af0a32d_batmon/install - The command '/bin/ash -o pipefail -c venv/bin/pip3 install -r requirements.txt' returned a non-zero code: 1

yuecelnabi, 2025-01-16:
Same problem, update doesn't work

MrPudding1, 2025-01-23:
I tried again today and it worked.

fl4p, 2025-02-02:
Hi, did you check http://homeassistant.local:8123//config/logs?provider=supervisor
supervisor logs for any more details?

novegit, 2025-04-11:
Updating 1.79 -> 1.82 fails. Error message:

Error updating Batmon: The command '/bin/ash -o pipefail -c python3 -m venv venv_bleak_pairing' returned a non-zero code: 1

Supervisor log:
2025-04-11 20:05:57.266 INFO (MainThread) [supervisor.docker.addon] Updating image 2af0a32d/amd64-addon-batmon:1.79 to 2af0a32d/amd64-addon-batmon:1.82
2025-04-11 20:05:57.295 INFO (MainThread) [supervisor.docker.addon] Starting build for 2af0a32d/amd64-addon-batmon:1.82
2025-04-11 20:06:01.831 ERROR (MainThread) [supervisor.docker.addon] Can't build 2af0a32d/amd64-addon-batmon:1.82: The command '/bin/ash -o pipefail -c python3 -m venv venv_bleak_pairing' returned a non-zero code: 
2025-04-11 20:06:01.832 ERROR (MainThread) [supervisor.docker.addon] Build log: 
Step 1/23 : ARG BUILD_FROM

Step 2/23 : FROM $BUILD_FROM

 ---> 02312a3ba918

Step 3/23 : WORKDIR /app

 ---> Using cache

 ---> 30a16e6726b2

Step 4/23 : RUN apk add python3

 ---> Using cache

 ---> be55e765a86f

Step 5/23 : RUN apk add bluez

 ---> Using cache

 ---> 69cd76b1e5d1

Step 6/23 : RUN apk add py-pip

 ---> Using cache

 ---> fff8afec126a

Step 7/23 : RUN apk add git

 ---> Using cache

 ---> 472f433fb1dd

Step 8/23 : COPY . .

 ---> Using cache

 ---> 18104cb5d679

Step 9/23 : RUN python3 -m venv venv_bleak_pairing

 ---> Running in 6fb2ea522a54

[Error: Command '['/app/venv_bleak_pairing/bin/python3', '-m', 'ensurepip', '--upgrade', '--default-pip']' returned non-zero exit status 1.
 ---> Removed intermediate container 6fb2ea522a54