fl4p/batmon-ha #272: no access to jbd BMS

RW-7, 2025-01-13:
Describe the issue here.
If you experience bluetooth connection problems, follow these steps before opening an issue:
https://github.com/fl4p/batmon-ha#troubleshooting
I entered the address directly in the config but also tried using the name.
It also recognizes the address using the name.
Unfortunately no access to the BMS.
On an ESP32 with source: github://syssi/esphome-jbd-bms@main the BMS is recognized and read.
But I would like to use battmon, it also saves the additional hardware.
Of course I deactivated the ESP before I started batmon
Raspberry Pi 4
Core
2025.1.2
Supervisor
2024.12.3
Operating System
14.1
Frontend
20250109.0
* Include exact BMS model name
* Query HW and SW numbers from the BMS app
* Enable verbose_log
JBD-SP04S034L4S200A  Version 0x44
```
Paste log output between BEGIN and END:
BEGIN 
09:56:33 INFO [bt] BT A4:C1:37:24:F3:33                      Links
09:56:33 INFO [bt] BT D5:41:0B:89:2E:65          D5-41-0B-89-2E-65
09:56:33 INFO [bt] BT A4:C1:37:34:F7:74                     Rechts
09:56:33 INFO [bt] BT 15:46:AA:9D:33:BB          15-46-AA-9D-33-BB
09:56:33 INFO [bt] BT A8:48:FA:1A:80:EA                classic300s
09:56:33 INFO [bt] BT A8:80:55:BD:99:79                         TY
09:56:33 ERROR [sampling] JbdBt(A4:C1:37:24:F3:33,bms0) device not found, retry in 1 seconds
09:56:33 DEBUG [sampling] retry in 1 sec
09:56:38 INFO [sampling] connecting bms JbdBt(A4:C1:37:24:F3:33,bms0)
09:56:38 INFO [bt] connecting bms0 (A4:C1:37:24:F3:33) adapter=default timeout=20
09:56:38 DEBUG [client] Connecting to device @ A4:C1:37:24:F3:33 with hci0
09:56:38 DEBUG [client] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_A4_C1_37_24_F3_33): ['org.bluez.Device1', {'Connected': <dbus_next.signature.Variant ('b', True)>}, []]
09:56:39 DEBUG [client] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_A4_C1_37_24_F3_33): ['org.bluez.Device1', {'Connected': <dbus_next.signature.Variant ('b', False)>}, []]
09:56:39 DEBUG [client] Device disconnected (/org/bluez/hci0/dev_A4_C1_37_24_F3_33)
09:56:39 DEBUG [client] _remove_signal_handlers(/org/bluez/hci0/dev_A4_C1_37_24_F3_33)
09:56:39 DEBUG [client] _disconnect_message_bus(/org/bluez/hci0/dev_A4_C1_37_24_F3_33)
09:56:39 ERROR [bt] [org.bluez.Error.Failed] le-connection-abort-by-local, starting scanner
09:56:39 INFO [bt] BT Discovery:
09:56:39 DEBUG [client] _remove_signal_handlers(/org/bluez/hci0/dev_A4_C1_37_24_F3_33)
09:56:39 DEBUG [client] no bus object
09:56:39 DEBUG [client] _disconnect_message_bus(/org/bluez/hci0/dev_A4_C1_37_24_F3_33)
09:56:39 DEBUG [client] already disconnected (/org/bluez/hci0/dev_A4_C1_37_24_F3_33)
END
```
[image]

skosko1977, 2025-01-18:
I have the same problem in version 1.8, on version 1.79 everything works ok.

erlendoyen, 2025-01-28:
i partly see the same issue with jbd.

it works for x number of hours, data can't be collected from one of the bms's

it's "resolved" by stopping batmon for a few minutes and starting it again, data is collected for both bms's

i have set the time interwall for collection to 60 sec and mqtt publishing to 65 seconds.
i have tested many values, makes no difference 

i have tested with and without keep alive
with and without bt_power_cycle

erlendoyen, 2025-01-29:
> I have the same problem in version 1.8, on version 1.79 everything works ok.

How did you downgrade when looking at the downgrade instructions I can't find the previous version

erlendoyen, 2025-01-30:
using these settings have increased the stability for me:

[image]

fl4p, 2025-02-02:
> > I have the same problem in version 1.8, on version 1.79 everything works ok.
> 
> How did you downgrade when looking at the downgrade instructions I can't find the previous version

@erlendoyen I've added v1.79 to https://github.com/fl4p/batmon-ha/tags