fl4p/batmon-ha #275: 1.81 update causes instability

kikuman, 2025-02-03:
Home Assistant says Battery becomes unavailable every minute after 1.81 update. Didn't have this issue with 1.80
Thank you!

ANT-16ZMB0TB140A
by ANT
Firmware: 16ZMUB00-220501A
Hardware: 16ZMB0TB140A

BEGIN 
[2af0a32d_batmon_2025-02-03T00-25-09.680Z.log](https://github.com/user-attachments/files/18634843/2af0a32d_batmon_2025-02-03T00-25-09.680Z.log)
END

fl4p, 2025-02-03:
did you try to downgrade back to 1.80 ?

cliftonarms, 2025-02-03:
I am also getting extensive device not found errors where v1.80 was very stable. How do I roll back to 1.8 ?

16:53:22 ERROR [sampling] BMS_8 error (#71): device 'dev_A5_C2_37_13_28_7B' not found
Traceback (most recent call last):
  File "/app/bmslib/sampling.py", line 155, in __call__
    s = await self._sample_inner()
        ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/bmslib/sampling.py", line 229, in _sample_inner
    async with bms:
               ^^^
  File "/app/bmslib/bt.py", line 344, in __aenter__
    await self.connect()
  File "/app/bmslib/models/jbd.py", line 51, in connect
    await super().connect(**kwargs)
  File "/app/bmslib/bt.py", line 236, in connect
    await self._connect_client(timeout=timeout)
  File "/app/bmslib/bt.py", line 185, in _connect_client
    await asyncio.wait_for(self.client.connect(timeout=timeout), timeout=timeout + 1)
  File "/usr/lib/python3.12/asyncio/tasks.py", line 520, in wait_for
    return await fut
           ^^^^^^^^^
  File "/app/venv/lib/python3.12/site-packages/bleak/__init__.py", line 615, in connect
    return await self._backend.connect(**kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/venv/lib/python3.12/site-packages/bleak/backends/bluezdbus/client.py", line 184, in connect
    watcher = manager.add_device_watcher(
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/venv/lib/python3.12/site-packages/bleak/backends/bluezdbus/manager.py", line 605, in add_device_watcher
    self._check_device(device_path)
  File "/app/venv/lib/python3.12/site-packages/bleak/backends/bluezdbus/manager.py", line 203, in _check_device
    raise BleakError(f"device '{device_path.split('/')[-1]}' not found")
bleak.exc.BleakError: device 'dev_A5_C2_37_13_28_7B' not found

oguzcav, 2025-02-04:
I was very happy with the new bleak, with 6 pieces of JK_PB2A16S20P s. I had similar issues with the older bleak. So, I'm skipping the update. I'm running 2025.1.4 on an i5 laptop, plain HA OS install, with an external BT dongle with 5 meters usb extension.

kikuman, 2025-02-04:
My ANT 16ZMB0TB140A are now working fine with 1.81
Thank you!

cliftonarms, 2025-02-04:
1.82 has run all night with no issues - thankyou.

fl4p, 2025-02-04:
> I was very happy with the new bleak, with 6 pieces of JK_PB2A16S20P s. I had similar issues with the older bleak. So, I'm skipping the update. I'm running 2025.1.4 on an i5 laptop, plain HA OS install, with an external BT dongle with 5 meters usb extension.

@oguzcav did you notice any difference with version 1.81 ?

@cliftonarms @kikuman thanks for the feedback!

ZonkoZoNk, 2025-03-23:
Taday i updated Home Assistant an now i have the same problems (unstable connection):
Thanks for Help

22:39:44 ERROR [sampling] Haus2 error (#2): timeout waiting for 2
Traceback (most recent call last):
  File "/usr/lib/python3.12/asyncio/tasks.py", line 520, in wait_for
    return await fut
           ^^^^^^^^^
asyncio.exceptions.CancelledError
The above exception was the direct cause of the following exception:
Traceback (most recent call last):
  File "/app/bmslib/__init__.py", line 81, in wait_for
    return await asyncio.wait_for(self._futures.get(name), timeout)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/asyncio/tasks.py", line 519, in wait_for
    async with timeouts.timeout(timeout):
               ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/asyncio/timeouts.py", line 115, in __aexit__
    raise TimeoutError from exc_val
TimeoutError
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
  File "/app/bmslib/sampling.py", line 155, in __call__
    s = await self._sample_inner()
        ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/bmslib/sampling.py", line 229, in _sample_inner
    async with bms:
               ^^^
  File "/app/bmslib/bt.py", line 348, in __aenter__
    await self.connect()
  File "/app/bmslib/models/jikong.py", line 153, in connect
    await self._q(cmd=0x96, resp=(0x02, 0x01))  # device state (resp 0x01 & 0x02)
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/bmslib/models/jikong.py", line 171, in _q
    return await self._fetch_futures.wait_for(resp, self.TIMEOUT)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/bmslib/__init__.py", line 75, in wait_for
    return await asyncio.gather(*tasks, return_exceptions=False)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/bmslib/__init__.py", line 84, in wait_for
    raise asyncio.TimeoutError("timeout waiting for %s" % name)
TimeoutError: timeout waiting for 2
22:39:44 INFO [sampling] Haus2 bms debug data: {'resp': {3: (bytearray(b'U\xaa\xeb\x90\x03"JK_B2A8S20P\x00\x00\x00\x00\x0011.XA\x00\x00\x0011.45\x00\x00\x00t\xf2>\x00\t\x00\x00\x00Haus2\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x001234\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00241211\x00\x00312212C0857\x00000\x00Input Userdata\x00\x00123456\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xfe\xf9\xff\xff\x1f-\x00\x02\x00\x00\x00\x00\x90\x1f\x00\x00\x00\x00\xc0\xd8\xe7\x00\x00\x00\x00\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x03\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00A\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xfe\xbf\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\xf0'), 1742765976.0652287), 1: (bytearray(b'U\xaa\xeb\x90\x01"\xd0\x07\x00\x00(\n\x00\x00Z\n\x00\x00B\x0e\x00\x00\xde\r\x00\x00\x05\x00\x00\x00\xe8\r\x00\x00<\n\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xc4\t\x00\x00\xa0\x86\x01\x00\x1e\x00\x00\x00<\x00\x00\x00\xa0\x86\x01\x00,\x01\x00\x00<\x00\x00\x00<\x00\x00\x00\xd0\x07\x00\x00\xbc\x02\x00\x00X\x02\x00\x00\xbc\x02\x00\x00X\x02\x00\x00\x00\x00\x00\x002\x00\x00\x00 \x03\x00\x00\xbc\x02\x00\x00\x08\x00\x00\x00\x01\x00\x00\x00\x01\x00\x00\x00\x01\x00\x00\x00\xc0E\x04\x00\xdc\x05\x00\x00\xb8\x0b\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x01\x00\x00\x00\x00\x00\x00\x00\xb0q\x0b\x00\x00\x00<2\x18\xfe\xf9\xff\xff\x1f)\x00\x02\x00\x00\x00\x00c'), 1742765979.5324924), 2: (bytearray(b'U\xaa\xeb\x90\x02\x19\x88\x0c\x8b\x0c\x8a\x0c\x8b\x0c\x8b\x0c\x8a\x0c\x8b\x0c\x8a\x0c\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xff\x00\x00\x00\x8a\x0c\x01\x00\x04\x007\x005\x004\x004\x004\x004\x004\x004\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xc0\x00\x00\x00\x00\x00Pd\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xa9\x00\xa7\x00\x00\x00\x08\x00\x00\x00\x00\t}c\x00\x00\xc0E\x04\x00\x0b\x00\x00\x00IB2\x00d\x00\x00\x00z\xf2>\x00\x01\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xff\x00\x01\x00\x00\x00\xc7\x03\x00\x00\x00\x00\x1d\x8d@@\x00\x00\x00\x00\x08\n\x00\x00\x00\x01\x00\x01\x00\x06\x00\x00\xea\xe7\x05\x00\x00\x00\x00\x00\xc0\x00\x00\x00\x00\x00\xca\x031\x97\x00\x001\x00\x00\x00\x80Q\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xfe\xff\x7f\xdc\x0f\x01\x00\x80\x03\x00\x00\x00B'), 1742765961.3826659)}, 'char_w': <bleak.backends.bluezdbus.characteristic.BleakGATTCha [... cut ...]
  File "/app/bmslib/sampling.py", line 155, in __call__
    s = await self._sample_inner()
        ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/bmslib/sampling.py", line 239, in _sample_inner
    sample = await bms.fetch()
             ^^^^^^^^^^^^^^^^^
  File "/app/bmslib/models/jikong.py", line 252, in fetch
    await self._fetch_futures.wait_for(0x02, self.TIMEOUT)
  File "/app/bmslib/__init__.py", line 84, in wait_for
    raise asyncio.TimeoutError("timeout waiting for %s" % name)
TimeoutError: timeout waiting for 2

fl4p, 2025-08-30:
can you try to reboot and see if it helps?