fl4p/batmon-ha #282: 1.82

matbogdan79, 2025-03-02:
After the last update to 1.82 my batman integration started to stop working or work intermittently.
It was working just fine in the previous versions and now this started to show up:
14:00:42 INFO [bt] BT Discovery:
14:00:47 INFO [bt] BT 5D:6F:53:6F:BB:53          5D-6F-53-6F-BB-53
14:00:47 INFO [bt] BT C8:47:80:07:58:B1               JK_B2A20S20P
14:00:47 INFO [main] Bleak version 0.20.2, BtBackend version bluez-v5.79
14:00:47 INFO [main] connecting mqtt homeassistant@core-mosquitto
14:00:47 INFO [main] Fetching 1 BMS + 0 virtual + 0 others concurrently, period=1.00s, keep_alive=True
14:00:47 INFO [sampling] connecting bms JKBt(C8:47:80:07:58:B1,battery1)
14:00:47 INFO [main] mqtt watchdog loop started with timeout 300.0s
14:00:54 INFO [jikong] battery1 normal connect failed (<class 'TimeoutError'>), connecting with scanner
14:01:06 INFO [sampling] connected bms JKBt(C8:47:80:07:58:B1,battery1)!
14:01:06 INFO [jikong] PSK = '0000' (Note that anyone within BLE range can read this!)
14:01:14 ERROR [sampling] battery1 error (#1): timeout waiting for 2
Traceback (most recent call last):
  File "/usr/lib/python3.12/asyncio/tasks.py", line 520, in wait_for
    return await fut
           ^^^^^^^^^
asyncio.exceptions.CancelledError
The above exception was the direct cause of the following exception:
Traceback (most recent call last):
  File "/app/bmslib/__init__.py", line 81, in wait_for
    return await asyncio.wait_for(self._futures.get(name), timeout)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/asyncio/tasks.py", line 519, in wait_for
    async with timeouts.timeout(timeout):
               ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/asyncio/timeouts.py", line 115, in __aexit__
    raise TimeoutError from exc_val
TimeoutError
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
  File "/app/bmslib/sampling.py", line 155, in __call__
    s = await self._sample_inner()
        ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/bmslib/sampling.py", line 239, in _sample_inner
    sample = await bms.fetch()
             ^^^^^^^^^^^^^^^^^
  File "/app/bmslib/models/jikong.py", line 252, in fetch
    await self._fetch_futures.wait_for(0x02, self.TIMEOUT)
  File "/app/bmslib/__init__.py", line 84, in wait_for
    raise asyncio.TimeoutError("timeout waiting for %s" % name)
TimeoutError: timeout waiting for 2
14:01:14 INFO [sampling] battery1 bms debug data: {'resp': {3: (bytearray(b'U\xaa\xeb\x90\x03^JK_B2A20S20P\x00\x00\x00\x0011.XW\x00\x00\x0011.288\x00\x00u$\xe5\x01\x08\x00\x00\x00JK_B2A20S20P\x00\x00\x00\x001234\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00240228\x00\x003091445561\x000000\x00Input Userdata\x00\x00111879\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00Input Userdata\x00\x00\xfc\xf9\xff\xff\x1f-\x00\x00\x00\x00\x00\x00\x90\x0f\x00\x00\x00\x00\xc0\xd8\x03\x00\x00\x00\x00\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xfe\x0f\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00I'), 1740916864.5472145), 1: (bytearray(b'U\xaa\xeb\x90\x01^X\x02\x00\x00Z\n\x00\x00_\n\x00\x00\x01\x0e\x00\x00\xf2\r\x00\x00\n\x00\x00\x00\xfc\r\x00\x00Z\n\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xc4\t\x00\x00\xf0I\x02\x00\x1e\x00\x00\x00\x1e\x00\x00\x00@\r\x03\x00\x1e\x00\x00\x00\x1e\x00\x00\x00\x1e\x00\x00\x00\xd0\x07\x00\x00\x86\x01\x00\x00|\x01\x00\x00\xae\x01\x00\x00\x90\x01\x00\x002\x00\x00\x00<\x00\x00\x00\xe8\x03\x00\x00 \x03\x00\x00\x10\x00\x00\x00\x01\x00\x00\x00\x01\x00\x00\x00\x01\x00\x00\x00 \xbf\x02\x00\xb8\x0b\x00\x00\x98\r\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xb0q\x0b\x00\x08\x00<2\x00\xfc\xf9\xff\xff\x1f-\x00\x00\x00\x00\x00\x00G'), 1740916865.8035328), 2: (bytearray(b'U\xaa\xeb\x90\x02^\x00\x0e\xff\r\xfc\r\x04\x0e\xfc\r\x04\x0e\xfc\r\xff\r\x00\x0e\xfb\r\x01\x0e\xfb\r\x02\x0e\xf6\r\xfd\r\xf6\r\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xff\xff\x00\x00\xfd\r\x06\x00\x01\x0f6\x006\x005\x007\x007\x00A\x00?\x004\x00<\x00;\x006\x007\x005\x005\x005\x008\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xb9\x00\x00\x00\x00\x00\xca\xdf\x00\x00\x0bx\x06\x00\xe8\x1c\x00\x00\x9d\x00n\x00\x00\x00\x00\x00\x00\x00\x00d \xbf\x02\x00 \xbf\x02\x00C\x00\x00\x00\x7f{\xba\x00d\x00\x00\x00x$\xe5\x01\x01\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xff\x00\x01\x00\x00\x00\xe8\x03%\x00\x07\x00\xe1\xc8B@\x00\x00\x00\x00a\x16\x0e\x1a\x00\x01\x00\x01\xc5\x05\x00\x00\x7f\xe7\x08\x0b\x00\x00\x00\x00\xb9\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xfe\xff\x7f\xdc\x0f\x01\x00\x00\x00\x00\x00\x00n'), 1740916866.496371)}, 'char_w': <bleak.backends.bluezdbus.characteristic.BleakGATTCharacteristicBlueZDBus object at 0x7f9b8f48f0>, 'char_r': <bleak.backends.bluezdbus.characteristic.BleakGATTCharacteristicBlueZDBus object at 0x7f9b8f48f0>}
14:01:14 INFO [sampling] battery1 device info: DeviceInfo(JK_B2A20S20P,hw-11.XW,sw-11.288,JK_B2A20S20P,#3091445561)
14:01:14 INFO [sampling] Bleak version 0.20.2
14:01:16 WARNING [bt] BMS JKBt(C8:47:80:07:58:B1,battery1) disconnected after 15.3s!
14:01:16 ERROR [sampling] battery1 error (#2): timeout waiting for 2
Traceback (most recent call last):
  File "/app/bmslib/__init__.py", line 81, in wait_for
    return await asyncio.wait_for(self._futures.get(name), timeout)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/asyncio/tasks.py", line 520, in wait_for
    return await fut
           ^^^^^^^^^
asyncio.exceptions.CancelledError
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
  File "/app/bmslib/sampling.py", line 155, in __call__
    s = await self._sample_inner()
        ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/bmslib/sampling.py", line 239, in _sample_inner
    sample = await bms.fetch()
             ^^^^^^^^^^^^^^^^^
  File "/app/bmslib/models/jikong.py", line 252, in fetch
    await self._fetch_futures.wait_for(0x02, self.TIMEOUT)
  File "/app/bmslib/__init__.py", line 84, in wait_for
    raise asyncio.TimeoutError("timeout waiting for %s" % name)
TimeoutError: timeout waiting for 2
14:01:16 INFO [sampling] battery1 bms debug data: {'resp': {3: (bytearray(b'U\xaa\xeb\x90\x03^JK_B2A20S20P\x00\x00\x00\x0011.XW\x00\x00\x0011.288\x00\x00u$\xe5\x01\x08\x00\x00\x00JK_B2A20S20P\x00\x00\x00\x001234\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00240228\x00\x003091445561\x000000\x00Input Userdata\x00\x00111879\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00Input Userdata\x00\x00\xfc\xf9\xff\xff\x1f-\x00\x00\x00\x00\x00\x00\x90\x0f\x00\x00\x00\x00\xc0\xd8\x03\x00\x00\x00\x00\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xfe\x0f\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00I'), 1740916864.5472145), 1: (bytearray(b'U\xaa\xeb\x90\x01^X\x02\x00\x00Z\n\x00\x00_\n\x00\x00\x01\x0e\x00\x00\xf2\r\x00\x00\n\x00\x00\x00\xfc\r\x00\x00Z\n\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xc4\t\x00\x00\xf0I\x02\x00\x1e\x00\x00\x00\x1e\x00\x00\x00@\r\x0 [... cut ...]
    await self.connect()
  File "/app/bmslib/models/jikong.py", line 152, in connect
    await self._q(cmd=0x97, resp=0x03)  # device info
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/bmslib/models/jikong.py", line 171, in _q
    return await self._fetch_futures.wait_for(resp, self.TIMEOUT)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/bmslib/__init__.py", line 84, in wait_for
    raise asyncio.TimeoutError("timeout waiting for %s" % name)
TimeoutError: timeout waiting for 3

fl4p, 2025-03-12:
did you try to downgrade to an earlier version?

chomats, 2025-03-25:
I see the same error in the logs. I fix it to restart module twice.

It's possible to increase the timeout by configuration ?

fl4p, 2025-08-30:
The timeout is currently set to 8 seconds (JK-bms).
This is usually enough and if we don't receive data within 8 seconds there is usually something wrong with the connection, so increasing this value would probably not help.

I guess this is an issue/bug with bleak and/or the BLE HW stack.