fl4p/batmon-ha #287: supervisor log warning: 'install_newer_bleak' does not exist in the schema for Batmon (2af0a32d_batmon)

typxxi, 2025-03-20:
JK inverter BMS V16 200 A - 2024 latest firmware

It is part of the supervisor log and I do not what to do. Integration is still working, but I have never heard about such issue 

```
home assistant 'install_newer_bleak' does not exist in the schema for Batmon (2af0a32d_batmon)
```

fl4p, 2025-03-27:
Just ignore. I removed the configuration parameter `install_newer_bleak` in a recent release, because it is not needed anymore. Hassio doesn't strip the value from existing configuration. This can be considered a bug in Home-Assistant

closed 2025-03-27 by fl4p

typxxi, 2025-04-21:
Hi,
is there anything I can do to get rid of the message it self cause it always pops up and quite often every 5 minutes.

[image]

mib1185, 2025-04-21:
it is documented [here](https://developers.home-assistant.io/docs/add-ons/configuration#options--schema) that:

> If you remove a configuration option from an add-on already deployed to users, it is recommended to delete the option to avoid a warning like Option '<options_key>' does not exist in the schema for <Add-on Name> (<add-on slug>).
> 
> To remove an option the Supervisor addons API can be used. Using bashio this boils down to bashio::addon.option '<options_key>' (without additional argument to delete this option key). Typically this should be called inside an if block checking if the option is still set using bashio::config.exists '<options_key>'.

so it is something, the add-on has to care about 😉

fl4p, 2025-05-29:
Ok, will fix this with the next release:

addon_main.sh:
```
if bashio::config.exists "install_newer_bleak"; then
  bashio::addon.option "install_newer_bleak"
fi
```