fl4p/batmon-ha #289: After setting up new Home Assistant, BattMon won't keep running for more than a few minutes, version 1.82

dhorsler, 2025-04-07:
After upgrading to new Home Assistant setup and reinstalling BattMon after last Raspberry Pi died a death, have been unable to get it to run for more than a few minutes before it crashes and dies.

its entirely possible that i've done something stupid and forgotten to do something that i had done previously to make it stable, but since it was several years ago, i can't remember what it was, if anything....

It appears to initially connect to BMS, take some readings and then lose connection to the BT BMS and then gets stuck in a retry loop which fails due to existing ongoing connection attempt.

Tried moving closer to BMS(s), no improvement.

It seems to indicate connection dififculties over BT LE, not sure why though as older setup had no such issues and config is the same.

After a while of trying this it kills the process on what appears to be:

./addon_main.sh: line 11:   104 Killed                  MQTT_HOST=$MQTT_HOST MQTT_USER=$MQTT_USER MQTT_PASSWORD=$MQTT_PASSWORD 

Have tried all the troubleshooting tips to no avail, used to work flawlessly on old Pi/HA setup.

Hardware: Raspberry Pi 3B using inbuilt Bluetooth
BMS: 4 x jbd (HW/SW versions to follow later....)
Batmon Version 1.82

Can restart it and it will connect to a couple of the BMS's and upload values via MQTT but then dies within 15 minutes each time.

older Pi was running an older version of Batmon, didn't update since it was working fine but now everything has upgraded to newer version including MQTT which was previously running a much older version and HA was some 2 years old before when it was stable.

have dropped the sampling/update periods right down in case that was causing an issue now reading every 3 to 5 minutes which should give sufficient time to sort itself out between read attempts/connections timeouts.

BEGIN 
bleak.exc.BleakDBusError: [org.bluez.Error.InProgress] Operation already in progress
07:36:58 INFO [sampling] Bleak version 0.20.2
07:36:58 INFO [sampling] connecting bms JbdBt(A4:C1:37:11:56:AC,inverter-2-battery-2)
07:36:58 INFO [bt] connecting inverter-2-battery-2 (A4:C1:37:11:56:AC) adapter=default timeout=20
07:36:58 DEBUG [client] Connecting to device @ A4:C1:37:11:56:AC
./addon_main.sh: line 11:   104 Killed                  MQTT_HOST=$MQTT_HOST MQTT_USER=$MQTT_USER MQTT_PASSWORD=$MQTT_PASSWORD /app/venv/bin/python3 main.py
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
07:37:47 INFO [main] Started in pair-only mode (bleak 0.20.2)
07:37:47 INFO [main] No PSK, nothing to pair
07:37:47 INFO [main] exit signal handler... (), {}, shutdown was False
07:37:49 INFO [bt] BT Discovery:
07:37:55 INFO [bt] BT Discovery:
07:38:01 INFO [bt] BT Discovery:
07:38:07 ERROR [main] Error discovering devices: [org.bluez.Error.InProgress] Operation already in progress
07:38:07 INFO [main] Verbose logging enabled
07:38:09 INFO [main] Bleak version 0.20.2, BtBackend version bluez-v5.79
07:38:09 INFO [main] connecting mqtt addons@core-mosquitto
07:38:10 DEBUG [client] Sending CONNECT (u1, p1, wr0, wq0, wf0, c1, k60) client_id=b''
07:38:10 INFO [main] Fetching 4 BMS + 0 virtual + 0 others serially, period=120.00s, keep_alive=False
07:38:10 INFO [sampling] connecting bms JbdBt(70:3e:97:D2:E0:3B,inverter-2-battery-1)
07:38:10 INFO [bt] connecting inverter-2-battery-1 (70:3e:97:D2:E0:3B) adapter=default timeout=20
07:38:10 DEBUG [client] Connecting to device @ 70:3e:97:D2:E0:3B
07:38:10 INFO [main] mqtt watchdog loop started with timeout 480.0s
07:38:11 DEBUG [client] Received CONNACK (0, 0)
07:38:30 ERROR [sampling] inverter-2-battery-1 error (#1): [org.bluez.Error.InProgress] Operation already in progress
Traceback (most recent call last):
  File "/app/bmslib/sampling.py", line 155, in __call__
    s = await self._sample_inner()
        ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/bmslib/sampling.py", line 229, in _sample_inner
    async with bms:
               ^^^
  File "/app/bmslib/bt.py", line 348, in __aenter__
    await self.connect()
  File "/app/bmslib/models/jbd.py", line 51, in connect
    await super().connect(**kwargs)
  File "/app/bmslib/bt.py", line 240, in connect
    await self._connect_client(timeout=timeout)
  File "/app/bmslib/bt.py", line 189, in _connect_client
    await asyncio.wait_for(self.client.connect(timeout=timeout), timeout=timeout + 1)
  File "/usr/lib/python3.12/asyncio/tasks.py", line 520, in wait_for
    return await fut
           ^^^^^^^^^
  File "/app/venv/lib/python3.12/site-packages/bleak/__init__.py", line 531, in connect
    return await self._backend.connect(**kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/venv/lib/python3.12/site-packages/bleak/backends/bluezdbus/client.py", line 124, in connect
    device = await BleakScanner.find_device_by_address(
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/venv/lib/python3.12/site-packages/bleak/__init__.py", line 310, in find_device_by_address
    return await cls.find_device_by_filter(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/venv/lib/python3.12/site-packages/bleak/__init__.py", line 369, in find_device_by_filter
    async with cls(detection_callback=apply_filter, **kwargs):
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/venv/lib/python3.12/site-packages/bleak/__init__.py", line 156, in __aexit__
    await self._backend.stop()
  File "/app/venv/lib/python3.12/site-packages/bleak/backends/bluezdbus/scanner.py", line 203, in stop
    await stop()
  File "/app/venv/lib/python3.12/site-packages/bleak/backends/bluezdbus/manager.py", line 390, in stop
    assert_reply(reply)
  File "/app/venv/lib/python3.12/site-packages/bleak/backends/bluezdbus/utils.py", line 22, in assert_reply
    raise BleakDBusError(reply.error_name, reply.body)
bleak.exc.BleakDBusError: [org.bluez.Error.InProgress] Operation already in progress
07:38:30 INFO [sampling] Bleak version 0.20.2
07:38:30 INFO [sampling] connecting bms JbdBt(70:3e:97:07:D8:17,inverter-1-battery-1)
07:38:30 INFO [bt] connecting inverter-1-battery-1 (70:3e:97:07:D8:17) adapter=default timeout=20
07:38:30 DEBUG [client] Connecting to device @ 70:3e:97:07:D8:17
07:38:50 ERROR [sampling] inverter-1-battery-1 error (#1): [org.bluez.Error.InProgress] Operation already in progress
Traceback (most recent call last):

END
```

genzo78, 2025-04-08:
Same exact situation as mine since yesterday

betmensk, 2025-06-25:
I had the same problem after updating Batmon to 1.82. 
HA core 2025.6.3 runs on RPi3B