fl4p/batmon-ha #295: ERROR in HA log: he requested state (392) exceeds the maximum allowed length (255).

typxxi, 2025-04-20:
I can not say when it started cause I have not checked those logs regulary.

I have a usual JK V15 inverter BMS with 200 A most recent HA (april.3) and Batmon-ha instaled
```
Cannot update state for entity sensor.jk_bms_discharge_message after processing payload on topic homeassistant/switch/battery1/discharge/config. The requested state (392) exceeds the maximum allowed length (255). Fall back to unknown, failed state: {"unique_id": "battery1__switch_discharge", "name": "discharge", "device_class": "outlet", "state_topic": "battery1/switch/discharge", "expire_after": 20.0, "device": {"identifiers": ["40110490690"], "manufacturer": "JK", "name": "HBB_PB2A16S20 (battery1)", "model": "JK_PB2A16S20P", "sw_version": "15.38", "hw_version": "15A"}, "command_topic": "homeassistant/switch/battery1/discharge/set"}
```
and here the full screenshot which might help too. It is just a warning but I wonder what is causing that and if I am the only one cause I can not remember what could have caused a 392 character and 137 character too long titel.

Sorry, I have not really much clue and my "batmon titles and name might be not the shortest, but for sure noch 137 charaters too long like: HBB_PB2A16S20 (battery1)

[image]

Thanks a lot for your great support !

typxxi, 2025-04-20:
And there is another similiar issues here, but this time not invalid state message but
can not update state - again 392 exceeds maximum length.

[image]

ilikenwf, 2025-07-02:
What I finally got working was disabling the watchddog checkbox on the configuration page. I also updated to bleak 1.0.1 but I'm not sure that really did anything.

fl4p, 2026-09-03:
`sensor.jk_bms_discharge_message` is not created by batmon; it is an MQTT sensor from your own HA configuration whose `state_topic` wildcard also matches `homeassistant/switch/battery1/discharge/config`, so it swallows the discovery JSON as a state. Narrow that topic (e.g. `battery1/switch/discharge`) or add a `value_template`, and the warning goes away. Closing since there is nothing to change in the add-on.

closed 2026-09-03 by fl4p