fl4p/batmon-ha #302: No Bluetooth Adaptor Found

9812ga98sd, 2025-06-27:
1. Home assistant running on a VM encountering no issues.
2. MQTT installed and working for a long time with other integrations.
3. CP2102 ESP32-WROOM-32D installed with ESPHome as a bluetooth proxy.
4. Batmon installed as an add on.
5. Bluetooth proxy sees all bluetooth devices and battery I want to connect to. Updates every 20 seconds or so.
Bluetooth proxy can see the mac address of the battery.
6. For some reason Batmon states "No Bluetooth adapters found."

* Include exact BMS model name
* Query HW and SW numbers from the BMS app
* Enable verbose_log

```
Paste log output between BEGIN and END:
BEGIN 
22:52:59 INFO [sampling] Bleak version 0.20.2
22:52:59 ERROR [main] 1 exceptions occurred fetching BMSs
22:52:59 ERROR [main] Error (num 201, max 200) reading BMS: No Bluetooth adapters found.
22:52:59 ERROR [main] Stack: Traceback (most recent call last):
  File "/app/main.py", line 37, in fetch_loop
    if await fn():
       ^^^^^^^^^^
  File "/app/main.py", line 331, in fn
    raise exceptions[0]
  File "/app/main.py", line 326, in fn
    await t()
  File "/app/bmslib/sampling.py", line 155, in __call__
    s = await self._sample_inner()
        ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/bmslib/sampling.py", line 229, in _sample_inner
    async with bms:
               ^^^
  File "/app/bmslib/bt.py", line 348, in __aenter__
    await self.connect()
  File "/app/bmslib/models/sok.py", line 106, in connect
    await super().connect(**kwargs)
  File "/app/bmslib/bt.py", line 240, in connect
    await self._connect_client(timeout=timeout)
  File "/app/bmslib/bt.py", line 189, in _connect_client
    await asyncio.wait_for(self.client.connect(timeout=timeout), timeout=timeout + 1)
  File "/usr/lib/python3.12/asyncio/tasks.py", line 520, in wait_for
    return await fut
           ^^^^^^^^^
  File "/app/venv/lib/python3.12/site-packages/bleak/__init__.py", line 531, in connect
    return await self._backend.connect(**kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/venv/lib/python3.12/site-packages/bleak/backends/bluezdbus/client.py", line 124, in connect
    device = await BleakScanner.find_device_by_address(
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/venv/lib/python3.12/site-packages/bleak/__init__.py", line 310, in find_device_by_address
    return await cls.find_device_by_filter(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/venv/lib/python3.12/site-packages/bleak/__init__.py", line 369, in find_device_by_filter
    async with cls(detection_callback=apply_filter, **kwargs):
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/venv/lib/python3.12/site-packages/bleak/__init__.py", line 152, in __aenter__
    await self._backend.start()
  File "/app/venv/lib/python3.12/site-packages/bleak/backends/bluezdbus/scanner.py", line 179, in start
    adapter_path = manager.get_default_adapter()
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/venv/lib/python3.12/site-packages/bleak/backends/bluezdbus/manager.py", line 299, in get_default_adapter
    raise BleakError("No Bluetooth adapters found.")
bleak.exc.BleakError: No Bluetooth adapters found.
22:52:59 WARNING [main] too many errors, abort
22:52:59 INFO [main] fetch_loop <function main.<locals>.fn at 0x7f5936da76a0> ends
22:52:59 INFO [main] All fetch loops ended. shutdown is already False
22:52:59 INFO [main] Disconnecting SokBt(D6:6C:0A:61:14:30,battery1)
22:52:59 INFO [main] exit signal handler... (), {}, shutdown was True
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped
END
```

jeremyjoslin, 2025-07-06:
> Reading up I can see that bluetooth proxies do not work 

I didn't know this. Why do you say this?
It's true that most proxies lack ability to do active proxy or GATT which is what seems to be necessary. I have an ESP32 device with active monitoring and GATT turned on, but get the same error (and same setup/situation).

Following along and interested if you figure it out.

9812ga98sd, 2025-07-07:
This was resolved by... changing the name of the battery. As I had changed the name of the battery I was advised to change it to begin with :
JDB-

Once I did that it was detected.

1. Install ESPHome via Home assistant web:  https://esphome.io/projects/?type=bluetooth 
2. Install ESPHome integration in Home assistant.
3. Install batmon add on
4. Make sure you can see your battery in the bluetooth scans here: http://YOUR Home assistant IP/config/bluetooth/advertisement-monitor
5. Rename battery to begin with JDB-
6. Batmon should see it.