fl4p/batmon-ha #304: ERROR: Got unexpected response from the API: Service not enabled

daubman, 2025-07-04:
BatMon won't start under the latest HA version on HA Green hardware:
Installation method: Home Assistant OS
Core: 2025.7.0
Supervisor: 2025.06.2
Operating System: 15.2
Frontend: 20250702.0

```
BEGIN 
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
[15:09:11] ERROR: Got unexpected response from the API: Service not enabled
[15:09:11] ERROR: Failed to get services from Supervisor API
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
[15:09:16] ERROR: Got unexpected response from the API: Service not enabled
[15:09:16] ERROR: Failed to get services from Supervisor API
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped
END
```

so4t, 2025-07-15:
That error means that MQTT add-on is not working.

argulp, 2025-07-17:
Hi, same issue for me, I'm not using MQTT addon... only a Mosquitto server outside the home assistant hardware