fl4p/batmon-ha #313: Error on connections after 1.84 update

giorgiopistola, 2025-09-05:
After the 1.84 the system disconnect and reconnect in a loop from both my devices, with this error:

BMK JK
SHOUNT VICTRON 

10:32:25 ERROR [main] 1 exceptions occurred fetching BMSs
10:32:25 ERROR [main] Error (num 96, max 200) reading BMS: 'BleakClient' object has no attribute 'get_services'
10:32:25 ERROR [main] Stack: Traceback (most recent call last):
  File "/app/main.py", line 37, in fetch_loop
    if await fn():
       ^^^^^^^^^^
  File "/app/main.py", line 346, in fn
    raise exceptions[0]
  File "/app/main.py", line 341, in fn
    await t()
  File "/app/bmslib/sampling.py", line 155, in __call__
    s = await self._sample_inner()
        ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/bmslib/sampling.py", line 239, in _sample_inner
    sample = await bms.fetch()
             ^^^^^^^^^^^^^^^^^
  File "/app/bmslib/models/victron.py", line 111, in fetch
    val = await self._fetch_value(k, reload_services=True)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/bmslib/models/victron.py", line 72, in _fetch_value
    await self.client.get_services()
          ^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'BleakClient' object has no attribute 'get_services'
```

lagwagone, 2025-09-05:
I'm also failing to connect to my smart shunt after upgrading: 

13:43:14 INFO [main] Batmon ver 1.84, Bleak ver 1.1.0, BtBackend ver bluez-v5.82
13:43:14 WARNING [bt] Installed bleak version 1.1.0 has no pairing agent, pairing with a pin will likely fail! 
13:43:14 INFO [main] connecting mqtt addons@core-mosquitto:1883
13:43:14 INFO [main] Fetching 1 BMS + 0 virtual + 0 others serially, period=2.00s, keep_alive=True
13:43:14 INFO [sampling] connecting bms SmartShuntBt(E8:2C:FB:4E:A0:CB,Must_HBP1800)
13:43:14 INFO [main] mqtt watchdog loop started with timeout 300.0s
13:43:22 ERROR [bt] Device with address E8:2C:FB:4E:A0:CB was not found., starting scanner
13:43:22 INFO [bt] BT Discovery:
13:43:27 INFO [bt] BT DD-EE-D2-BB-DD-37                       None
13:43:27 INFO [bt] BT 9D-CA-5C-0B-EF-9F                         None
13:43:27 ERROR [sampling] SmartShuntBt(E8:2C:FB:4E:A0:CB,Must_HBP1800) device not found, retry in 7 seconds (Device with address E8:2C:FB:4E:A0:CB was not found.)

lagwagone, 2025-09-06:
Same problem here - also Victron SmartShunt

07:39:01 INFO [sampling] Must_HBP1800 device info: DeviceInfo(SmartShunt,hw-None,sw-None)
07:39:01 INFO [sampling] Bleak version 1.1.0
07:39:01 ERROR [main] 1 exceptions occurred fetching BMSs
07:39:01 ERROR [main] Error (num 4, max 200) reading BMS: 'BleakClient' object has no attribute 'get_services'
07:39:01 ERROR [main] Stack: Traceback (most recent call last):
  File "/app/main.py", line 37, in fetch_loop
    if await fn():
       ^^^^^^^^^^
  File "/app/main.py", line 346, in fn
    raise exceptions[0]
  File "/app/main.py", line 341, in fn
    await t()
  File "/app/bmslib/sampling.py", line 155, in __call__
    s = await self._sample_inner()
        ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/bmslib/sampling.py", line 239, in _sample_inner
    sample = await bms.fetch()
             ^^^^^^^^^^^^^^^^^
  File "/app/bmslib/models/victron.py", line 111, in fetch
    val = await self._fetch_value(k, reload_services=True)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/bmslib/models/victron.py", line 72, in _fetch_value
    await self.client.get_services()
          ^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'BleakClient' object has no attribute 'get_services'
07:39:04 ERROR [sampling] Must_HBP1800 error (#5): 'BleakClient' object has no attribute 'get_services'
Traceback (most recent call last):
  File "/app/bmslib/sampling.py", line 155, in __call__
    s = await self._sample_inner()
        ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/bmslib/sampling.py", line 239, in _sample_inner
    sample = await bms.fetch()
             ^^^^^^^^^^^^^^^^^
  File "/app/bmslib/models/victron.py", line 111, in fetch
    val = await self._fetch_value(k, reload_services=True)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/bmslib/models/victron.py", line 72, in _fetch_value
    await self.client.get_services()
          ^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'BleakClient' object has no attribute 'get_services'

fl4p, 2025-09-06:
please try to install https://github.com/fl4p/batmon-ha/archive/refs/heads/master.zip

lagwagone, 2025-09-07:
Took me a minute how to install this locally, for those who don't know extract the repo to config/addons and then select check for updates on the add-on store page and it should appear under local add-ons. 

@fl4p new alpha is working 100%

closed 2025-09-08 by fl4p