fl4p/batmon-ha #314: After upgrade from 1.83 to 1.84 service wont start

atticushome, 2025-09-05:
I am using custom MQTT broker without username and password.
Here is the yaml file:
`devices:
  - address: C8:47:80:16:AA:E4
    type: jk_32s
    alias: battery00
    adapter: hci0
  - address: C8:47:80:16:AB:04
    type: jk_32s
    alias: battery01
    adapter: hci0
concurrent_sampling: true
keep_alive: true
verbose_log: true
sample_period: 5
publish_period: 7
invert_current: true
watchdog: true
expire_values_after: 60
bt_power_cycle: true
install_newer_bleak: true
mqtt_broker: 192.168.188.249
mqtt_port: 1883
mqtt_user: ""
mqtt_password: ""
`
And here is the error I am getting:
MQTT service not configured in HA. Using broker credentials from add-on configuration.
18:55:17 INFO [main] Started in pair-only mode (bleak 0.13.1a1)
18:55:17 INFO [main] No PSK, nothing to pair
18:55:17 INFO [main] exit signal handler... (), {}, shutdown was False
./addon_main.sh: line 28: MQTT_HOST: unbound variable
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped

Confushion, 2025-09-05:
Same issue here.
Looks like when using external MQTT broker (outside hass supervisors control) the configured MQTT url is not correctly initialised.

SSonder0815, 2025-09-05:
same issue and log structure with me as well. Needed to restore old version.

fl4p, 2025-09-06:
Will fix this with next update.
HA bashio environments appear to vary a lot in behavior depending on platform, version, etc ..
This didn't happen in my test environment where I use the latest updates.