fl4p/batmon-ha #321: number of temperature sensors

erlendoyen, 2025-09-22:
Hi

are there any way to adjust the number of battery sensor's that it will populate?

i have a bms with 3 sensors, but i have 253 temperature sensors populated

[image]

erlendoyen, 2025-09-30:
hi, i don't have the full name of the bms available right now

and i also see that device item page for the bms displays only the temperatures sensors with data.

it's only in the HA overview page where the number's of sensors for the device goes up to 255.

mariusfl20, 2025-09-30:
Not sure what happened but mine is looping in the following error :
"TypeError: type NoneType doesn't define __round__ method"

BMS type - JK_B2A20S20P

First AI check indicates that my sensors are sending "null data" -  I know for sure temp data is not live (at least for the external sensors)

when i try to connect to my JK BMS using BATMON in home assistant i get the following error:

...........part error log........ 

19:38:16 ERROR [main] 1 exceptions occurred fetching BMSs

19:38:16 ERROR [main] Error (num 5, max 200) reading BMS: type NoneType doesn't define __round__ method

19:38:16 ERROR [main] Stack: Traceback (most recent call last):

  File "/app/main.py", line 37, in fetch_loop

    if await fn():

       ^^^^^^^^^^

  File "/app/main.py", line 350, in fn

    raise exceptions[0]

  File "/app/main.py", line 345, in fn

    await t()

  File "/app/bmslib/sampling.py", line 155, in __call__

    s = await self._sample_inner()

        ^^^^^^^^^^^^^^^^^^^^^^^^^^

  File "/app/bmslib/sampling.py", line 263, in _sample_inner

    sample.temperatures = self._filter_temperatures(sample.temperatures)

                          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

  File "/app/bmslib/sampling.py", line 206, in _filter_temperatures

    return [round(self._lhq_temp[i].add(temperatures[i]), 2) for i in range(len(temperatures))]

            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

TypeError: type NoneType doesn't define __round__ method

19:38:19 ERROR [sampling] Baterie18KWh error (#7): type NoneType doesn't define __round__ method

Traceback (most recent call last):

  File "/app/bmslib/sampling.py", line 155, in __call__

    s = await self._sample_inner()

        ^^^^^^^^^^^^^^^^^^^^^^^^^^

  File "/app/bmslib/sampling.py", line 263, in _sample_inner

    sample.temperatures = self._filter_temperatures(sample.temperatures)

                          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

  File "/app/bmslib/sampling.py", line 206, in _filter_temperatures

    return [round(self._lhq_temp[i].add(temperatures[i]), 2) for i in range(len(temperatures))]

            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

TypeError: type NoneType doesn't define __round__ method

19:38:19 INFO [sampling] Baterie18KWh bms debug data: {'resp': {3:

fl4p, 2026-09-03:
This was a corrupt frame from the BMS declaring 253 NTC sensors; batmon decoded past the end of the payload and HA discovery created an entity per phantom slot. Frames are checksum-validated since 2.14 and the count byte is now bounds-checked on master, so it cannot recur. The stale entities you already deleted stay gone. Closing.

closed 2026-09-03 by fl4p