fl4p/batmon-ha #323: Docker build failed for 2af0a32d/aarch64-addon-batmon:1.87 (exit code 1)

Ferrari2, 2025-09-29:
Describe the issue here.
I get this issue when I try to install batmon-ha on my working HA system...

```
Paste log output between BEGIN and END:
BEGIN 
Docker build failed for 2af0a32d/aarch64-addon-batmon:1.87 (exit code 1). Build output: #0 building with "default" instance using docker driver #1 [internal] load build definition from Dockerfile #1 transferring dockerfile: 963B 0.0s done #1 WARN: InvalidDefaultArgInFrom: Default value for ARG $BUILD_FROM results in empty or invalid base image name (line 2) #1 DONE 0.1s #2 [internal] load metadata for ghcr.io/home-assistant/aarch64-base:latest #2 DONE 0.3s #3 [internal] load .dockerignore #3 transferring context: 2B done #3 DONE 0.1s #4 [ 1/15] FROM ghcr.io/home-assistant/aarch64-base:latest@sha256:f03c7ea28e11c627f2eb08ff2ac24faa14ef80c50d173193fdb43675a12f282a #4 DONE 0.0s #5 [ 2/15] WORKDIR /app #5 CACHED #6 [internal] load build context #6 transferring context: 659.70kB 0.3s done #6 DONE 0.4s #7 [ 3/15] RUN apk add python3~3.12 || apk add python3 #7 0.788 fetch https://dl-cdn.alpinelinux.org/alpine/v3.22/main/aarch64/APKINDEX.tar.gz #7 5.792 WARNING: updating and opening https://dl-cdn.alpinelinux.org/alpine/v3.22/main: temporary error (try again later) #7 5.792 fetch https://dl-cdn.alpinelinux.org/alpine/v3.22/community/aarch64/APKINDEX.tar.gz #7 10.80 WARNING: updating and opening https://dl-cdn.alpinelinux.org/alpine/v3.22/community: temporary error (try again later) #7 10.80 ERROR: unable to select packages: #7 10.80 python3 (no such package): #7 10.80 required by: world[python3~3.12] #7 10.84 fetch https://dl-cdn.alpinelinux.org/alpine/v3.22/main/aarch64/APKINDEX.tar.gz #7 15.84 fetch https://dl-cdn.alpinelinux.org/alpine/v3.22/community/aarch64/APKINDEX.tar.gz #7 15.84 WARNING: updating and opening https://dl-cdn.alpinelinux.org/alpine/v3.22/main: temporary error (try again later) #7 20.85 WARNING: updating and opening https://dl-cdn.alpinelinux.org/alpine/v3.22/community: temporary error (try again later) #7 20.85 ERROR: unable to select packages: #7 20.85 python3 (no such package): #7 20.85 required by: world[python3] #7 ERROR: process "/bin/ash -o pipefail -c apk add python3~3.12 || apk add python3" did not complete successfully: exit code: 1 ------ > [ 3/15] RUN apk add python3~3.12 || apk add python3: 10.80 ERROR: unable to select packages: 10.80 python3 (no such package): 10.80 required by: world[python3~3.12] 10.84 fetch https://dl-cdn.alpinelinux.org/alpine/v3.22/main/aarch64/APKINDEX.tar.gz 15.84 fetch https://dl-cdn.alpinelinux.org/alpine/v3.22/community/aarch64/APKINDEX.tar.gz 15.84 WARNING: updating and opening https://dl-cdn.alpinelinux.org/alpine/v3.22/main: temporary error (try again later) 20.85 WARNING: updating and opening https://dl-cdn.alpinelinux.org/alpine/v3.22/community: temporary error (try again later) 20.85 ERROR: unable to select packages: 20.85 python3 (no such package): 20.85 required by: world[python3] ------ [33m1 warning found (use docker --debug to expand): [0m - InvalidDefaultArgInFrom: Default value for ARG $BUILD_FROM results in empty or invalid base image name (line 2) Dockerfile:10 -------------------- 8 | # RUN apk add --no-cache python3 bluez py-pip git 9 | 10 | >>> RUN apk add python3~3.12 || apk add python3 11 | RUN apk add bluez 12 | #RUN apk add bluez < 5.66-r4" -------------------- ERROR: failed to build: failed to solve: process "/bin/ash -o pipefail -c apk add python3~3.12 || apk add python3" did not complete successfully: exit code: 1
END
```

bas0008, 2025-09-30:
I have the same problem with RPI5.

Docker build failed for 2af0a32d/aarch64-addon-batmon:1.88 (exit code 1). Build output: #0 building with "default" instance using docker driver #1 [internal] load build definition from Dockerfile #1 transferring dockerfile: 963B done #1 WARN: InvalidDefaultArgInFrom: Default value for ARG $BUILD_FROM results in empty or invalid base image name (line 2) #1 DONE 0.1s #2 [internal] load metadata for ghcr.io/home-assistant/aarch64-base:latest #2 DONE 0.2s #3 [internal] load .dockerignore #3 transferring context: 2B done #3 DONE 0.1s #4 [ 1/15] FROM ghcr.io/home-assistant/aarch64-base:latest@sha256:f03c7ea28e11c627f2eb08ff2ac24faa14ef80c50d173193fdb43675a12f282a #4 DONE 0.0s #5 [ 2/15] WORKDIR /app #5 CACHED #6 [internal] load build context #6 transferring context: 664.95kB 0.1s done #6 DONE 0.2s #7 [ 3/15] RUN apk add python3~3.12 || apk add python3 #7 0.389 fetch https://dl-cdn.alpinelinux.org/alpine/v3.22/main/aarch64/APKINDEX.tar.gz #7 5.395 fetch https://dl-cdn.alpinelinux.org/alpine/v3.22/community/aarch64/APKINDEX.tar.gz #7 5.395 WARNING: updating and opening https://dl-cdn.alpinelinux.org/alpine/v3.22/main: temporary error (try again later) #7 10.40 WARNING: updating and opening https://dl-cdn.alpinelinux.org/alpine/v3.22/community: temporary error (try again later) #7 10.40 ERROR: unable to select packages: #7 10.40 python3 (no such package): #7 10.40 required by: world[python3~3.12] #7 10.41 fetch https://dl-cdn.alpinelinux.org/alpine/v3.22/main/aarch64/APKINDEX.tar.gz #7 15.41 fetch https://dl-cdn.alpinelinux.org/alpine/v3.22/community/aarch64/APKINDEX.tar.gz #7 15.41 WARNING: updating and opening https://dl-cdn.alpinelinux.org/alpine/v3.22/main: temporary error (try again later) #7 20.42 WARNING: updating and opening https://dl-cdn.alpinelinux.org/alpine/v3.22/community: temporary error (try again later) #7 20.42 ERROR: unable to select packages: #7 20.42 python3 (no such package): #7 20.42 required by: world[python3] #7 ERROR: process "/bin/ash -o pipefail -c apk add python3~3.12 || apk add python3" did not complete successfully: exit code: 1 ------ > [ 3/15] RUN apk add python3~3.12 || apk add python3: 10.40 ERROR: unable to select packages: 10.40 python3 (no such package): 10.40 required by: world[python3~3.12] 10.41 fetch https://dl-cdn.alpinelinux.org/alpine/v3.22/main/aarch64/APKINDEX.tar.gz 15.41 fetch https://dl-cdn.alpinelinux.org/alpine/v3.22/community/aarch64/APKINDEX.tar.gz 15.41 WARNING: updating and opening https://dl-cdn.alpinelinux.org/alpine/v3.22/main: temporary error (try again later) 20.42 WARNING: updating and opening https://dl-cdn.alpinelinux.org/alpine/v3.22/community: temporary error (try again later) 20.42 ERROR: unable to select packages: 20.42 python3 (no such package): 20.42 required by: world[python3] ------ [33m1 warning found (use docker --debug to expand): [0m - InvalidDefaultArgInFrom: Default value for ARG $BUILD_FROM results in empty or invalid base image name (line 2) Dockerfile:10 -------------------- 8 | # RUN apk add --no-cache python3 bluez py-pip git 9 | 10 | >>> RUN apk add python3~3.12 || apk add python3 11 | RUN apk add bluez 12 | #RUN apk add bluez < 5.66-r4" -------------------- ERROR: failed to build: failed to solve: process "/bin/ash -o pipefail -c apk add python3~3.12 || apk add python3" did not complete successfully: exit code: 1

fl4p, 2025-12-27:
as mentioned in the error message, this could be a temporary problem. try again the next day

closed 2026-05-27 by fl4p